OctoPrint/OctoPrint #1077: UnicodeDecodeError

xflowx, 2015-10-01:
After upgrading to version 1.2.6., everytime i start a print i get the following error: 

UnicodeDecodeError: ''ascii' codec can't decode byte 0xef in position 0: ordinal not in range(128)' @ comm.py:startPrint:600

seems to have something to do with the encoding.
with the old version(>1.2) these gcode files worked.
Also, new uploaded files wont work, everytime the same error

xflowx, 2015-10-01:
Printer: K8200
<script src="https://gist.github.com/xflowx/743f6076430822fc61ae.js">Terminal</script>

Log says nothing about this error

xflowx, 2015-10-01:
terminal output:

Send: M105
Recv: ok T:22.0 /0.0 B:21.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:22.0 /0.0 B:21.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:22.0 /0.0 B:21.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:22.0 /0.0 B:21.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:22.0 /0.0 B:21.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:22.0 /0.0 B:21.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:22.0 /0.0 B:21.0 /0.0 @:0 B@:0
Changing monitoring state from 'Operational' to 'Printing'
Changing monitoring state from 'Printing' to 'Error: UnicodeDecodeError: ''ascii' codec can't decode byte 0xef in position 0: ordinal not in range(128)' @ comm.py:startPrint:600'

foosel, 2015-10-02:
Please provide an example GCODE file with which that can be reproduced. Not copy pasted, really upload the exact file and link it here.

Also always provide the log. There is more information about your setup in there, not just about the error, and it also happened in the past that people didn't provide it, told me there was nothing in it, then when they finally did provide it it turned out there was a hint in there they didn't recognize as such. There is a reason the bug reporting template asks for this, untruncated.

xflowx, 2015-10-02:
here is the log file:
http://outerspace-astronomy.de/octoprint/octoprint.log
and here one example gcode
http://outerspace-astronomy.de/octoprint/top_raspi_logo.gcode

this file printed well before upgrading!

foosel, 2015-10-02:
> this file printed well before upgrading!

Yes, you already said that, no need to get agitated :) 

You also stated that you had upgraded from a version ">1.2" when in fact the log shows that the last version before 1.2.6 was 1.1.1, so in fact "<1.2", which is very important information when trying to figure out why it started to produce errors after you upgraded...

The error is caused by [UTF-8 BOMs](https://en.wikipedia.org/wiki/Byte_order_mark) in your files. They all start with a magic number basically. Your old version of OctoPrint still had a bug that basically sent all lines as is from the file, without sanitizing them first. The printer communication protocol actually states that only ASCII encoding is allowed, so this was wrong and got fixed in 1.2. It however hiccups hard on a BOM which it should just ignore (although a BOM shouldn't be in that file in the first place, it should not be encoded in UTF-8...). I will fix that in the code, you should however definitely try to figure out why your slicer is producing BOMed files - I have not seen this in the wild so far, certainly not during my own excessive testing across operating systems, so something is off here in your workflow.

xflowx, 2015-10-02:
I also already thought that it has something to do with BOM's, i also fell a few times over them.. ;-)

The gcode comes out directly from Repetier using Slicer, with the "save gcode" icon used....

I'll find out where the BOM's come from and will keep you informed!
Thanks for the quick answer on that!

foosel, 2015-10-05:
Fixed in `devel` and `maintenance`, will  be part of 1.2.7 release

closed 2015-10-05 by foosel