OctoPrint/OctoPrint #1078: urllib timeout capturing timelapse images from Blue Iris (with patch)

krisp, 2015-10-02:
I am using a popular CCTV package called Blue Iris to manage cameras around my property, so I also have the IP camera watching the 3d printer hooked into it so I can monitor along with everything else. To that end, I have had no success getting the OctoPrint timelapse feature to work downloading still images from the correct URL. I could view the jpeg in the browser and wget and curl the jpeg on the octoprint vm. However, timelapse.py would always throw a urllib timeout error when trying to pull a snapshot.

I spent a few minutes debugging this with urllib and httprequest debugging enabled and found that urllib.urlretrieve() would connect to blue iris' http server but hang forever retrieving the image. Since this behavior is different than curl and wget I figured it must be something with urllib and tested using python's "requests" library instead which works fine with Blue Iris and should work with any other web server.

This patch changes timelapse.py from urllib to requests and writes jpegs in 1kb chunks rather than storing the whole image in ram.

```python
--- a/timelapse.py        2015-09-01 10:42:50.205005558 -0400
+++ b/timelapse.py       2015-10-01 22:12:25.853462979 -0400
@@ -6,7 +6,7 @@
 import logging
 import os
 import threading
-import urllib
+import requests
 import time
 import subprocess
 import fnmatch
@@ -251,7 +251,13 @@
        def _captureWorker(self, filename):
                eventManager().fire(Events.CAPTURE_START, {"file": filename})
                try:
-                       urllib.urlretrieve(self._snapshotUrl, filename)
+                       r = requests.get(self._snapshotUrl, stream=True)
+                       with open(filename, "wb") as fd:
+                         for chunk in r.iter_content(chunk_size=1024):
+                           if chunk:
+                             fd.write(chunk)
+                             fd.flush()
+
                        self._logger.debug("Image %s captured from %s" % (filename, self._snapshotUrl))
                except:
                        self._logger.exception("Could not capture image %s from %s, decreasing image counter again" % (filename, self._snapshotUrl))
```

foosel, 2015-10-02:
I already did that on July 20th on the `devel` branch (see https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/timelapse.py#L302) - the code is even pretty much identical. I'll look into including the change into the `maintenance` branch however so it will be included into 1.2.7 and your webcam will work with the current release versions as well, not just the development version :)

foosel, 2015-10-03:
Ported to `maintenancen` branch, will be included in 1.2.7 release.

closed 2015-10-03 by foosel

krisp, 2015-10-03:
super, thanks!