OctoPrint/OctoPrint #1079: Terminal and custom control failed to send gcode (G28, M302, for example)

mateddy, 2015-10-02:
Enter "G28" or "M302" in terminal tab and hit "Send".  Alternatively, I have a custom control "G29" for Kossel bed autoprobe in "control" tab.

In case of G28/G29, my Kossel should home/autoprobe.  In case of M302, should acknowledge and "OK". However, nothing happened instead.

Octoprint 1.3.0.dev490+g874ebde (devel branch), Kossel Mini running Marlin, Firefox in OSX El Capitan, Safari in iOS 9.  Problem not observed in last devel branch (just updated 2 hours ago)

However, clicking "X/Y Home", "Z Home" or loading gcode file are all fine.  Printer working properly and accepting command using e.g. Repetier Host.

foosel, 2015-10-02:
Fixed.

> Problem not observed in last devel branch (just updated 2 hours ago)

This however I do not understand. The problem was introduced yesterday when I merged the `dev/clientlib` branch, so it should definitely have be observable for you (it certainly was for me on the same version).

closed 2015-10-02 by foosel

mateddy, 2015-10-02:
Thank you very much!

commit 5644b5dc (2015-10-02): clientlib: Fixed some refactoring errors still referring to exports var

Closes #1079