OctoPrint/OctoPrint #1080: [REQUEST] Add printing status to page title....

bolsoncerrado, 2015-10-05:
...so when page is minimized, rolling over it we can read what's going on (ie. printing, operational, disconnected, etc.)

foosel, 2015-10-05:
That sounds like a perfect candidate for a plugin, no need to add that to the core.

foosel, 2015-10-05:
Implementation:

  - [ ] Client side plugin, so only implement `AssetPlugin` mixin and have OctoPrint inject a custom view model, unbound to anything
  - [ ] Viewmodel should have `printerStateViewModel` injected and subscribe to `printerStateViewModel.stateString`
  - [ ] On change of state string, adjust title

foosel, 2015-10-05:
That was a mistaken click if there ever was one... sorry for the noise.

myersCody, 2016-02-22:
Hello,
My partner and I would like to take part in this particular issue. However, we believe this issue as been solved at:

http://plugins.octoprint.org/plugins/title_status/

We are curious if we should continue our work, or move on to a new issue.

Thanks,

Cody Myers & Angela Reinhart

foosel, 2016-02-22:
Argh, indeed it was, thanks for the heads up

closed 2016-02-22 by foosel