OctoPrint/OctoPrint #1082: [REQUEST] Nativly support an On/Off Power Supply Toggle

Grogyan, 2015-10-07:
Currently using OctoPrint 1.2.6

As I am just about to have my 3D printers setup to automatically turn off the ATX power supply at the end of a print and looking at adding a custom control toggle to enable the user turn on/off the power supply via an M80/M81.

Looking at what OctoPrint currently has as native/stock commands in the Control tab, there is IMO, utilized Fan On and Fan Off buttons.

I feel that these, or one of them could be used instead as a Power Supply Toggle button.
Best used with Red/Green colour scheme to indicate to the user, whether the Power Supply is On/Off

In addition, looking at the documentation I could see no mention of an actual Toggle control button, which would eliminate excessive buttons.

markwal, 2015-10-08:
Toggles are a bummer when talking to a machine over a serial line because the current state is not *known* for a similar reason toggles on IR remote controls are a bummer.

You can add these controls to the control tab yourself pretty easily.  Either via the guide in the documentation or via the custom control plugin that allows editing the buttons in settings.

http://docs.octoprint.org/en/master/features/custom_controls.html

and 

http://plugins.octoprint.org/plugins/customControl/

The key to that plugin is to right click on stuff.

But perhaps you knew about this already?  Since you said Natively.   However, I think it is better in the customization because ATX power supplies aren't that common yet.

Grogyan, 2015-10-08:
Lol, that is why the plugin didn't work for me.

In all seriousness, I believe that an On/Off toggle should be implemented within the Control tab, maybe even replacing the unused Fan On, Fan Off buttons.
Anyone can easily create a custom control for doing so, however it shouldn't be a custom control, but one that is there, natively, with Stock OctoPrint/OctoPi Image.

ATX power supplies for 3D printers are quite common, especially for RepRaps, though this isn't the only method to initiate a M80/M81  command to be able to turn on or off a power supply, just that  ATX power supplies make this feature quite simple to implement electrically

My general feeling is that it should be there, for saftety/good practice

foosel, 2015-10-08:
Pretty much every printer sold our built has a hot end fan controlled via M106. Way way less do have an ATX power supply (I'd go out on a limb here and saggy those are actually declining too). Some that do control it via the firmware, done don't but have it instead weird to some pin some place else. OctoPrint is extendable to be able to be adjusted to the user's individual setup for a reason - it is completely impossible to make it a one size fits all tool, and it simply makes no sense to add something like a PSU button set to the set of default controls if it only is useful for a small percentage of users that way. Tasks like this are *exactly* why there are custom controls which are even extendable through plugins, and why there are system commands and even semi automatic things possible once you add plugins into the mix (i have one of my printers wired so it switches the relay to the printer on when I try to connect to it and switching it off when it is disconnected or when nobody is watching the UI and no print is active after a minute - that was a cookie of lines and would be absolutely misplaced in the core application since it is so specific to my hardware setup). 

So no, not going to implement this in the core application, it simply makes no sense for most users. Those that can profit from it can easily add a couple of lines to a config file to get it and if sufficient demand exists I'm sure someone will also write a plugin.

Also: What Mark said about toggle buttons holds true, they necessitate a way to determine the current state on a "cold start", we don't have that in the context of stuff like fan control or the ATX commands.

closed 2015-10-08 by foosel