OctoPrint/OctoPrint #1083: SD disabled = error

MoonshineSG, 2015-10-12:
Trying to start Octoprint (raspberry pi) with "feature"->"sdSupport: false" will cause fatal UI JS error:

"Uncaught TypeError: Cannot read property 'fileupload' of undefined" in "packed_app.js" (source static/js/app/viewmodels/files.js)
```
function setDropzone(dropzone, enable) {
                var button = (dropzone == "local") ? self.uploadButton : self.uploadSdButton;
                var drop = (dropzone == "local") ? localTarget : sdTarget;
                var url = API_BASEURL + "files/" + dropzone;

                button.fileupload({
                    url: url,
                    dataType: "json",
                    dropZone: enable ? drop : null,
                    drop: function(e, data) {

                    },
                    done: gcode_upload_done,
                    fail: gcode_upload_fail,
                    progressall: gcode_upload_progress
                });
```

With sdSupport set to true, everything is ok.

Salandora, 2015-10-12:
Right now I pushed a fix for this.

MoonshineSG, 2015-10-12:
works.

closed 2015-10-12 by MoonshineSG