OctoPrint/OctoPrint #1084: Dividers

MoonshineSG, 2015-10-12:
What were you doing?
add dividers into the system menu

What did you expect to happen?
menu to be split in 3 sections

```
system:
  actions:
  - action: printer on
    command: gpio -g write 24 1
    name: Turn printer ON
  - action: camera on
    command: gpio -g write 32 1;gpio -g write 23 1
    name: Turn IR lights ON
  - action: divider
  - action: printer off
    command: gpio -g write 24 0
    confirm: You are about to stop the printer.
    name: Turn printer OFF ☂
  - action: camera off
    command: gpio -g write 32 0;gpio -g write 23 0
    name: Turn IR lights OFF
  - action: divider
  - action: shutdown
    command: sudo shutdown -h now
    confirm: You are about to shutdown the system.
    name: Shutdown ☂
  - action: reboot
    command: sudo shutdown -r now
    confirm: You are about to reboot the system
    name: Reboot  ☂
  - action: restart
    command: sudo service octoprint restart
    confirm: You are about to restart OctoPrint
    name: Restart OctoPrint  ☂
```

What happened instead?
only first "- action: divider" works

[image]

Branch & Commit or Version of OctoPrint:
latest (and some prev versions)

Printer model & used firmware incl. version
(if applicable - always include if unsure):
irelevant

Browser and Version of Browser, Operating 
System running Browser (if applicable - always 
include if unsure):
safari/chrome, but would fail on any...

Link to octoprint.log on gist.github.com or pastebin.com
(ALWAYS INCLUDE AND DO NOT TRUNCATE):
n/a

Link to contents of terminal tab or serial.log on 
gist.github.com or pastebin.com (if applicable - always 
include if unsure or reporting communication issues AND
DO NOT TRUNCATE):
n/a

Link to contents of Javascript console in the browser 
on gist.github.com or pastebin.com or alternatively a
screenshot (if applicable - always include if unsure
or reporting UI issues):
n/a

Screenshot(s) showing the problem (if applicable - always
include if unsure or reporting UI issues):

I have read the FAQ.

Note: this used to work, but I am not sure when it stopped working...

foosel, 2015-10-12:
> Branch & Commit or Version of OctoPrint:
> latest (and some prev versions)

latest what? devel? some feature branch? please just provide the version string as displayed on the ui, it saves us a lot of guessing :)

MoonshineSG, 2015-10-12:
sorry abut that,,  Version: 1.3.0.dev530+g1bf46d6 (HEAD branch)

Salandora, 2015-10-12:
This one here is tricky, I found the problem, but because of 

   specs[spec["action"]] = copied

it is not that easy to solve, because spec["action"] = "divider"  for all dividers :-)
possible solution divider0, divider1, ... what do you think @foosel?

foosel, 2015-10-12:
Ugly ;) Has to wait for a week then, we'll need to brainstorm about this
done more.

foosel, 2015-10-19:
The above commit should do it. Each divider gets a custom id for use in the `specs` dictionary. That id never gets communicated outside of the application (it's really only used as key for the `specs` dict), but this way we don't have any overwriting issues any more:

[image]

MoonshineSG, 2015-10-19:
yes. tested and looks good.

Thanks.

commit 4cfc74c3 (2015-10-19): Fix: Each divider action needs a custom action id

Closes #1084
closed 2015-10-19 by MoonshineSG