OctoPrint/OctoPrint #1085: custom control param

MoonshineSG, 2015-10-12:
What were you doing?
use custom control with parameters

What did you expect to happen?
parameter to be replace by actual value

```
[...]   
    - commands:
      - M851 Z-%(distance)
      - M500
      - M501
      input:
      - default: 0
        name: Set offset
        parameter: distance
        slider: false
```

[image]

What happened instead?
parameter name used instead

[image]

Branch & Commit or Version of OctoPrint:
latest (and some prev versions)

Printer model & used firmware incl. version
(if applicable - always include if unsure):
irrelevant

Browser and Version of Browser, Operating 
System running Browser (if applicable - always 
include if unsure):
irrelevant

Link to octoprint.log on gist.github.com or pastebin.com
(ALWAYS INCLUDE AND DO NOT TRUNCATE):
n/a

Link to contents of terminal tab or serial.log on 
gist.github.com or pastebin.com (if applicable - always 
include if unsure or reporting communication issues AND
DO NOT TRUNCATE):
n/a

Link to contents of Javascript console in the browser 
on gist.github.com or pastebin.com or alternatively a
screenshot (if applicable - always include if unsure
or reporting UI issues):
no error gets reported

Screenshot(s) showing the problem (if applicable - always
include if unsure or reporting UI issues):
see above

I have read the FAQ.

foosel, 2015-10-12:
Your parameter format is missing the actual format string after the parameter name. What happens if you try "%(distance)d"?

MoonshineSG, 2015-10-12:
Sorry, the cat ate it !!   :D 

But doesn't help... 
[image]

foosel, 2015-10-12:
@Salandora could you take a look at this? Or it will have to wait until next week.

Salandora, 2015-10-12:
Interesting. Will take a look.

Salandora, 2015-10-12:
Pushed a commit that fixes the problem.

MoonshineSG, 2015-10-12:
works. but with `%(distance)s`

otherwise it gives "TypeError: %d format: a number is required, not unicode"

MoonshineSG, 2015-10-12:
not sure if this is related.... I tested a custom control that calls a script which contains parameters... 

```
    - input:
      - default: X
        name: Axis
        parameter: axis
        slider: false
      - default: 10
        name: Distance
        parameter: test_length
        slider:
          max: 20
          min: 1
      - default: 5
        name: Repeat
        parameter: repetitions
        slider:
          max: 20
          min: 2
      name: Go
      script: custom/repeatability.gcode
```

~/.octoprint/scripts/gcode/custom/repeatability.gcode
```
G91

{% for n in range(parameters.repetitions) %}
  G1 {{ parameters.axis }}{{ parameters.test_length }} F1000
  G4 S1
  G1 {{ parameters.axis }}-{{ parameters.test_length }} F1000
  G4 S1
{% endfor %}

G90
```

JS console throws a HTTP 404 error with message "Unknown script: custom/repeatability.gcode"

Again, this used to work....

Salandora, 2015-10-12:
Another interesting problem. Will check later this day.

foosel, 2015-10-12:
Can you provide info on which version you last saw it working? That would
help narrow it down.

MoonshineSG, 2015-10-13:
sorry, but i cant tell for sure. I don't use this custom control often so I don't know when it stopped working. But it used to work, thats for sure.

Salandora, 2015-10-13:
@foosel alright I checked the GCODE Script problem. 
It got introduced trough: https://github.com/foosel/OctoPrint/commit/bbd728c51bad7e24a2adca424030fc3281ac4285

Inside there you suddenly changed the behaviour of the script command, so it only accepts a name and a context, but the parameters/inputs don't get used anymore.

foosel, 2015-10-19:
So a migration error it was. Well, it *is* a development branch. While the goal of course is to keep that in working shape at all times as well, stuff CAN break. Should be fixed now however :)

MoonshineSG, 2015-10-19:
tested and seams fixed. 

It was not a complain. Just pointing out minor gaps... we all understand it's development branch and we accept the risk.

closed 2015-10-19 by MoonshineSG

foosel, 2015-10-19:
> we all understand it's development branch

I fear not all do, but thanks that you do :)

MoonshineSG, 2015-10-19:
There's a saying in my country: "there is no forest without brushwood" (english equivalent: "no garden without weeds")

commit 66fbd1fc (2015-10-19): Fix: Also provide custom control parameters to custom GCODE scripts

Closes #1085