OctoPrint/OctoPrint #1086: Sending G Code M42 Using The Terminal to Ultimaker 2 (Marlin)

scottdanesi, 2015-10-12:
Hello Everyone,
I seem to be experiencing an issue when sending the M42 command using OctoPrint to my Ultimaker 2 machine.  This command is used to turn on and off the LEDs on the machine.  When I send the command through the OctoPrint Terminal, the connection will lock up and no longer accept any other commands.  I can disconnect and reconnect manually and the connection comes back alive.

I have also opened up a thread on the Ultimaker forums to verify that this is the correct command for the Marlin firmware on the Ultimaker 2.

https://ultimaker.com/en/community/view/17756-control-ultimaker-2-leds-via-gcode?page=unread

Thoughts?

Thank you,
--Scott

scottdanesi, 2015-10-12:
Ticket Info:
What was I doing? I was in the Terminal window with the printer sitting idle and I sent the command M42 S0 to the Ultimaker 2.

What did I expect? I expected the LEDs on the Ultimaker 2 to shut off and the connection to stay active.

What happened? The connection to the Ultimaker 2 locked up and would not send a response that the command was received. Also the M105 requests stopped.

Version: OctoPrint 1.2.6

Printer and Firmware: Ultimaker 2 with v15.02.1

Log: http://pastebin.com/YMTx3Edm

Contents of Terminal Window: 
Send: M105
Recv: ok T:30.3 /0.0 B:51.6 /0.0 @:0 B@:0
Send: M105
Recv: ok T:30.6 /0.0 B:51.0 /0.0 @:0 B@:0
Send: M42 S0

Thank you,
--Scott

Salandora, 2015-10-12:
Hello,

did you tried to use the command with another Host software? Repetier-Host or Printrun?
Normally the connection shouldn't lock up just because you send a command.

scottdanesi, 2015-10-12:
I agree.  I have not used any other software to test out this command.  Is there anyone that has OctoPrint set up with an Ultimaker 2 that can test to be sure it is not just an issue with my machine?

I will try and send the command using Printrun to see if it is having the same issue and I will report back.  

Thank you,
--Scott Danesi

scottdanesi, 2015-10-12:
Hi Everyone,
I just tried to send the command using Printrun and it failed the same way that OctoPrint did.  Looks like we have an issue with the machine.  I apologize for the false alarm on this one.

Thank you,
--Scott Danesi

closed 2015-10-12 by scottdanesi