OctoPrint/OctoPrint #1088: Communication hung up on Percent Sign

dancole, 2015-10-13:
The gcode generators, e.g. Fusion 360,  I use wrap the code in percent signs. OctoPrint hangs up on this. Can we change something about this? Either a setting or just ignore that line or something.

I'm using Smoothieboard hardware btw, maybe their not ack it correctly?

dancole, 2015-10-13:
Terminal output: http://pastebin.com/xTBXnDcb

markwal, 2015-10-13:
Are you sure you got a gcode file out of Fusion 360?  Upload a gcode of a simple model somewhere and paste a link here.

dancole, 2015-10-13:
It exports as an NC extension. I selected the Generic Smoothie Mill Configuration (They have a grbl configuration as well). Here's the file: http://pastebin.com/3RDiiPqF

The code makes 3 loops around, cutting out a cube, and then faces the top. I'm running a CNC router. I have run the pasted code, but I have to press pause/resume a couple of times to get past the percent sign, tool change, and motor start.

markwal, 2015-10-13:
Huh. Pretty cool.  I didn't know it could do that.  I wonder why it wraps it in '%'.  Haven't seen any gcode with that before. You can, of course, edit out the '%' with a text editor and you could make a plugin that eats them on the way to the printer.

According to https://en.wikipedia.org/wiki/G-code % means start of data (and end) during file transfer.  Originally used to stop tape rewind.  I guess it'd be ok to just toss them out though some software that emits them might intend for content before and after them to be ignored which would be harder to implement.

foosel, 2015-10-19:
@dancole I'm a bit surprised that the communication stalls. OctoPrint by default ships with a setting to not expect wait for `ok`s for commands that are not recognized as GCODE (which the `%`) is. Could you check if you have `feature > unknownCommandsNeedAck` set to `true` in your `config.yaml`? If so, try setting it to `false`.

foosel, 2015-10-19:
Also: https://github.com/OctoPrint/OctoPrint-Commfilter

dancole, 2015-10-23:
There was no `unknownCommandsNeedAck` line. I created one and set it to false, then restarted OctoPrint, but it still timed-out on %. The plugin worked great though, adding `^%` removed the issue.

foosel, 2015-11-05:
Closing this then since the plugin solves the problem.

closed 2015-11-05 by foosel