OctoPrint/OctoPrint #1089: Automatic start Octoprint.

krzysiekd, 2015-10-13:
I managed to implement Octopi on Raspberry pi B+ but I have small problem.
I find a script for Automatic start Octoprint.
sudo cp scripts/octoprint.init /etc/init.d/octoprint
sudo chmod +x /etc/init.d/octoprint
sudo cp scripts/octoprint.default /etc/default/octoprint
Of course it's a script for Octoprint,not for Octopi (I understand that pi version it's minimized normal version). It's possible to use this script in Octopi ?

markwal, 2015-10-13:
Shouldn't need it on octopi.  Octopi already sets up octoprint to autostart when the pi boots up.

closed 2015-10-13 by markwal

rhuuk420, 2015-10-19:
this issue needs to reviewed. I am building from scratch using the online instructions for Octoprint. I can reproduce the fact that the instructions for building your own appear to fail when you apply the auto startup scripts. I could not find a way to apply logging.

rhuuk420, 2015-10-19:
What were you doing?
installing octoprint on latest raspbian OS. when applying the auto start service for init.d and default configs, applying the venv string in both files, I still could not get the octprint service to start. no logging is available to turn on to identify what is the root cause issue in /etc/init.d/octoprint  . the /etc/default/octoprint file contains the pi user account, and the right path to /home/pi/OctoPrint/venv/bin/octoprint. 

2. What did you expect to happen?
service to start up. logs to be available to tell me why it failed, or some info at the cmd line when I start the service as to why it is failing. 

3. What happened instead?
no log information, no PID running, no octoprint service running un root, or pi, or anything.

4. Branch & Commit or Version of OctoPrint:

latest from git clone as per instructions. 

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
not applicable - basic OS raspbian config for autostart is not working. 

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
not applicable. raspbian OS daemon not starting. 

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):
log does not contain any changes when this occurs. 

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):
both files are 0 bytes. 

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):
not applicable. 
10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):
not applicable.

I have read the FAQ.
I have searched the issues. this is closest and most relevant.

mjleffler, 2015-11-05:
I was having a similar problem but came to the realization that I hadn't un-commented `DAEMON=/home/pi/OctoPrint/venv/bin/octoprint` in `/etc/default/octoprint`. Thought it'd be nice if I wasn't the only one making that oversight.