OctoPrint/OctoPrint #1090: Octoprint won't compile on windows 10 x86

Ryotsuke, 2015-10-15:
Windows 10 x86
Python v2.7.10

Writing c:\users\overco~1\appdata\local\temp\easy_install-ap99es\netifaces-0.8\setup.cfg
Running netifaces-0.8\setup.py -q bdist_egg --dist-dir c:\users\overco~1\appdata\local\temp\easy_install-ap99es\netifaces-0.8\egg-dist-tmp-nd8n5_
netifaces.c
netifaces.c(406) : error C2275: 'PyObject' : illegal use of this type as an expression
        c:\python27\include\object.h(108) : see declaration of 'PyObject'
netifaces.c(406) : error C2065: 'py_family' : undeclared identifier
netifaces.c(407) : error C2275: 'PyObject' : illegal use of this type as an expression
        c:\python27\include\object.h(108) : see declaration of 'PyObject'
netifaces.c(407) : error C2065: 'list' : undeclared identifier
netifaces.c(407) : error C2065: 'py_family' : undeclared identifier
netifaces.c(407) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(407) : warning C4024: 'PyDict_GetItem' : different types for formal and actual parameter 2
netifaces.c(409) : error C2065: 'py_family' : undeclared identifier
netifaces.c(411) : error C2065: 'list' : undeclared identifier
netifaces.c(411) : warning C4047: '==' : 'int' differs in levels of indirection from 'void *'
netifaces.c(411) : error C2065: 'list' : undeclared identifier
netifaces.c(411) : error C2065: 'list' : undeclared identifier
netifaces.c(411) : error C2065: 'list' : undeclared identifier
netifaces.c(415) : error C2065: 'list' : undeclared identifier
netifaces.c(416) : error C2065: 'list' : undeclared identifier
netifaces.c(416) : warning C4047: '=' : 'int' differs in levels of indirection from 'PyObject *'
netifaces.c(417) : error C2065: 'list' : undeclared identifier
netifaces.c(419) : error C2065: 'py_family' : undeclared identifier
netifaces.c(419) : error C2065: 'py_family' : undeclared identifier
netifaces.c(419) : error C2065: 'py_family' : undeclared identifier
netifaces.c(423) : error C2065: 'list' : undeclared identifier
netifaces.c(424) : error C2065: 'py_family' : undeclared identifier
netifaces.c(424) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(424) : warning C4024: 'PyDict_SetItem' : different types for formal and actual parameter 2
netifaces.c(424) : error C2065: 'list' : undeclared identifier
netifaces.c(424) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(424) : warning C4024: 'PyDict_SetItem' : different types for formal and actual parameter 3
netifaces.c(425) : error C2065: 'list' : undeclared identifier
netifaces.c(425) : error C2065: 'list' : undeclared identifier
netifaces.c(425) : error C2065: 'list' : undeclared identifier
netifaces.c(427) : error C2065: 'list' : undeclared identifier
netifaces.c(427) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(427) : warning C4024: 'PyList_Append' : different types for formal and actual parameter 1
netifaces.c(716) : error C2275: 'PyObject' : illegal use of this type as an expression
        c:\python27\include\object.h(108) : see declaration of 'PyObject'
netifaces.c(716) : error C2065: 'dict' : undeclared identifier
netifaces.c(718) : error C2065: 'dict' : undeclared identifier
netifaces.c(718) : warning C4047: '=' : 'int' differs in levels of indirection from 'PyObject *'
netifaces.c(720) : error C2065: 'dict' : undeclared identifier
netifaces.c(730) : error C2065: 'dict' : undeclared identifier
netifaces.c(730) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(730) : warning C4024: 'PyDict_SetItemString' : different types for formal and actual parameter 1
netifaces.c(732) : error C2065: 'dict' : undeclared identifier
netifaces.c(732) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(732) : warning C4024: 'PyDict_SetItemString' : different types for formal and actual parameter 1
netifaces.c(734) : error C2065: 'dict' : undeclared identifier
netifaces.c(734) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(734) : warning C4024: 'PyDict_SetItemString' : different types for formal and actual parameter 1
netifaces.c(740) : error C2065: 'dict' : undeclared identifier
netifaces.c(740) : warning C4047: 'function' : 'PyObject *' differs in levels of indirection from 'int'
netifaces.c(740) : warning C4024: 'add_to_family' : different types for formal and actual parameter 3
error: Setup script exited with error: command 'C:\\Users\\OverColoredCat\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\cl.exe' failed with exit status 2

foosel, 2015-10-19:
You manually installed netifaces 0.10, the stable version of OctoPrint still requires netifaces 0.8. It tried then to compile that version since the installed one didn't match and ran into trouble since you don't have compilation of binary libraries setup in your windows install.

I changed the required version for the `maintenance` branch (future `master`) now to 0.10 (see the above commit), that will be released with 1.2.7 within the next days.

Ryotsuke, 2015-10-19:
I believe if specific version is required, https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows should say that

foosel, 2015-10-19:
It is a wiki :)

foosel, 2015-11-04:
1.2.7 has been released for some days now and I just updated the [Setup Guide for Windows](https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows), closing this.

closed 2015-11-04 by foosel