OctoPrint/OctoPrint #1091: Mystifying threading error amidst Cura slicing

chrisvoncsefalvay, 2015-10-18:
So, I tried to slice a file with the Cura plugin for the first time and got the following error (I suspect this might not necessarily be a Cura specific issue, hence raising it here):

```
2015-10-18 10:20:40,456 - octoprint.plugins.cura - INFO - Running u'/home/pi/CuraEngine/build/ -v -p -s [blah blah my settings profile INI file was echoed here] -o /tmp/tmpu6CqLY.gco /home/pi/.octoprint/uploads/wide-short-metric.stl' in /home/pi/CuraEngine/build
Exception in thread Thread-21:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 1136, in run_node
    result = getattr(self, method)(node, input, async)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 1282, in run_command_node
    node.cmd.run(input=input, async=async)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 632, in run
    self.process = p = Popen(self.args, **self.kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 679, in __init__
    errread, errwrite)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 545, in _execute_child
    super(Popen, self)._execute_child(args, executable, preexec, *rest)
  File "/usr/lib/python2.7/subprocess.py", line 1259, in _execute_child
    raise child_exception
OSError: [Errno 13] Permission denied
```
Seems to be some quirky Sarge/multithreading issue. Has this occurred to anyone?

(RPi running OctoPrint 1.2.6, master)

Salandora, 2015-10-18:
At a first look I would say the path to your CuraEngine is wrong.

Try to Add the name of the Binary file into the path so that it e.g. look like this:
/home/pi/CuraEngine/build/CuraEngine

markwal, 2015-10-18:
I'm sure that's it, sarge is trying to run '/home/pi/CuraEngine/build/' which is a folder, not an executable.

closed 2015-10-18 by markwal

chrisvoncsefalvay, 2015-11-02:
Indeed. Duh :) Silly me. Thanks for fixing this so quickly ;)