OctoPrint/OctoPrint #1093: Octoprint inaccessible after router reset

l2smith2, 2015-10-19:
Octoprint web UI no longer accessible after a network fault. e.g. the wifi router the raspberry pi connects to is rebooted.

Occasionally after long periods of printer idle time connection is also unsuccessful.

Hard rebooting the pi solves this. Inconvenient during long prints

nophead, 2015-10-19:
My guess would be the router has lost the DNS record for it but it would still be accessible by its ip address.

l2smith2, 2015-10-19:
It is only ever accessible via its IP address. I have checked the router client list and the ip hasn't changed since last connection

nophead, 2015-10-19:
Can you ping it?
Can you SSH into it?

l2smith2, 2015-10-19:
I will try again on the next occurrence. But usually when the browser won't connect to the UI there is no  response trying to ssh the pi. It will continue to print without issue, just no control. 

Sometimes during long prints If I haven't logged into the web UI for some time it can take 5-15 mins to connect and not just say page not found

Salandora, 2015-10-20:
Well first. If your network crashes or your router reboots and your pi isn't accessible through ping or SSH then it is not octoprints fault. It is the fault of your pi not reconnecting to the network.

l2smith2, 2015-10-20:
What is the best way to solve that then?

Sometimes I have wil access via the octodroid mobile app but accessing from
the local network isn't possible

Salandora, 2015-10-20:
I would try to give your pi a static IP address. If your use a wired solution.

If your use a wireless LAN solution it could also help to reconnect the WLAN Adapter.
This way your force the connection to be reestablished.

l2smith2, 2015-10-20:
I use the edimax wireless adapter. Is there a way to make octoprint or the
pi periodically check for a network connection and attempt a reconnect or
reset the adapter if not present?

markwal, 2015-10-20:
@l2smith2, perhaps this thread: https://www.raspberrypi.org/forums/viewtopic.php?t=16054 is useful?

foosel, 2015-10-20:
This has nothing to do with OctoPrint, so this is definitely the wrong location. OctoPrint is a userspace application that has nothing to do with how the host system it's running on does or doesn't get connectivity.

The thread @markwal linked is relevant. As is - in case you are running OctoPi (the SD card image) - [this ticket in its bug tracker](https://github.com/guysoft/OctoPi/issues/145).

closed 2015-10-20 by foosel