OctoPrint/OctoPrint #1094: raspbian instruction for autostart daemon not working.

rhuuk420, 2015-10-19:
What were you doing?

installing OctoPrint on fresh raspbian OS. When applying the auto start service for init.d and default configs, applying the venv string in the default file, I still could not get the octprint service to start. 

no logging switch is clearly marked or available to turn on to identify what is the root cause issue in /etc/init.d/octoprint . 

The /etc/default/octoprint file contains the pi user account, and the right path to /home/pi/OctoPrint/venv/bin/octoprint. 

messages log file does not show anything. auth.log file shows root attempting to open the service. see the following as the only evidence I can find:

Oct 19 23:42:49 raspberrypi sudo:       pi : TTY=pts/1 ; PWD=/ ; USER=root ; COMMAND=/usr/sbin/service OctoPrint start
Oct 19 23:42:49 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Oct 19 23:42:50 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Oct 19 23:42:56 raspberrypi sudo:       pi : TTY=pts/1 ; PWD=/ ; USER=root ; COMMAND=/usr/sbin/service octoprint start
Oct 19 23:42:56 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Oct 19 23:42:56 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Oct 19 23:44:11 raspberrypi sudo:       pi : TTY=pts/1 ; PWD=/var/log ; USER=root ; COMMAND=/usr/sbin/service octoprint start
Oct 19 23:44:11 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Oct 19 23:44:11 raspberrypi sudo: pam_unix(sudo:session): session closed for user root 

1.What did you expect to happen?

service to start up. logs to be available to tell me how or why it failed, or some echo info at the cmd line when I start the service as to why it is failing. 

2.What happened instead?
 no log information, no PID running, no octoprint service running un root, or pi, or anything.

3.Branch & Commit or Version of OctoPrint:

latest from git clone as per instructions. 

1.Printer model & used firmware incl. version
 (if applicable - always include if unsure):
 
not applicable - basic OS raspbian config for autostart is not working. not gotten that far.

2.Browser and Version of Browser, Operating 
 System running Browser (if applicable - always 
 include if unsure):
 not applicable. raspbian OS daemon not starting. 

3.Link to octoprint.log on gist.github.com or pastebin.com
 (ALWAYS INCLUDE AND DO NOT TRUNCATE):
 log does not contain any changes when this occurs. 

4.Link to contents of terminal tab or serial.log on 
 gist.github.com or pastebin.com (if applicable - always 
 include if unsure or reporting communication issues AND
 DO NOT TRUNCATE):
 both files are 0 bytes. 

5.Link to contents of Javascript console in the browser 
 on gist.github.com or pastebin.com or alternatively a
 screenshot (if applicable - always include if unsure
 or reporting UI issues):
 not applicable. 

6.Screenshot(s) showing the problem (if applicable - always include if unsure or reporting UI issues): not applicable.

I have read the FAQ.
 I have searched the issues. this is closest and most relevant.

Salandora, 2015-10-20:
Pleasee attach the /etc/default/octoprint and /etc/init.d/octoprint files.
Also which guide did you use (if any) to set octoprint up?

foosel, 2015-10-20:
What did it output when you ran 

    sudo update-rc.d octoprint defaults

?

rhuuk420, 2015-10-20:
i posted my config files to a secret gist here at https://gist.github.com/rhuuk420/2f4befbfcc0fd2c3c8dd since i was unable to attach files to the issue, per write perms. 

I know enough about linux to be dangerous, but not enough to hack through this code so i can help only so much (windows guy 8-) ) 

when I run the sudo command as instructed I receive:
update-rc.d: using dependency based boot sequencing

i ran this before I posted the gist files.

rhuuk420, 2015-10-20:
Also, I confirmed Octoprint runs fine when called from the cmd line. it just does not start with the autodaemon. 
I've run through those instruction again on a 2nd pi this morning, with the same effect. 

Linux raspberrypi 4.1.10-v7+ #821 SMP PREEMPT Sat Oct 10 00:16:28 BST 2015 armv7l GNU/Linux

rhuuk420, 2015-10-20:
as for which instructions, I was using the ones available here:
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian

rhuuk420, 2015-10-20:
interesting... running the command from the main page
octoprint --daemon {start|stop|restart} [--pid PIDFILE] 

ps aux shows:
pi       26622  2.2  2.9  96904 25688 ?        Sl   17:28   0:08 /home/pi/OctoPrint/venv/bin/python ./octoprint --daemon start

puts it in run mode, but again, from the command line. 
service --status-all
shows the following:  [ ? ]  octoprint

sudo service octoprint start

will not start the service. I wonder of suspect it might have something to do with the venv? or the userID to runas?

restarted the device, and the octoprint service shows [?], and there are no logs in octoprint for the daemon. but, i can confirm I can run it as a daemon from the prompt as pi. 

hope this helps.

Salandora, 2015-10-20:
Yup as expected: inside your /etc/default/octoprint file you need to remove the "#" in front of

    #DAEMON=/home/pi/OctoPrint/venv/bin/octoprint

so it looks like this:

    DAEMON=/home/pi/OctoPrint/venv/bin/octoprint

That should fix your problem..

rhuuk420, 2015-10-21:
sigh. I either need new eyes, or a bigger screen. I feel like i'm back in high school with that kind of a miss.  

thx

closed 2015-10-21 by rhuuk420

glyndavidson, 2017-03-31:
Doh! Thank you Salandora!