OctoPrint/OctoPrint #1095: Custom controls to run command?

apemanzilla, 2015-10-21:
I was wondering if it was possible to have a custom control that, instead of sending gcode to the printer, would run a command on the host. I use a Raspberry Pi, and have LEDs connected to the GPIO. I have to SSH to it and run a script to toggle the lights on/off, and it would be nice to be able to add a button to the control tab to run the script for me. Is this possible in the current version?

CapnBry, 2015-10-21:
It is in the documentation, create an EventHook with type "system".
http://docs.octoprint.org/en/master/events/index.html

apemanzilla, 2015-10-21:
@CapnBry I have read that, but there is no way to trigger an event hook from a button AFAIK. The end goal is to have a button in the control tab of the web interface labelled "Lights On" or similar, that, when clicked, runs a command or shell script to turn on the LEDs.

CapnBry, 2015-10-21:
Oh I am sorry you did say a custom control. It looks like Custom Controls can only do Printer commands (and tool commands?) but you could always add a system command to the top menu to allow you to run your lights off/on commands
http://docs.octoprint.org/en/master/configuration/config_yaml.html#system

apemanzilla, 2015-10-21:
Alright, thanks, I'll try that.

Looking at the event hooks, it mentions events for when M80 and M81 are sent. My printer does not use those, so I could try adding controls for M80 and M81 and then adding event hooks that toggle lights on/off for M80 and M81 respectively.

closed 2015-10-21 by apemanzilla

fraternl, 2015-10-27:
In that example it says "gpio ...." and that works most of the time, but sometimes it didn't...
I even got an error message. Rebooting the system worked.

I investigated this and changed the line to "sudo gpio ......" and now it works all the time.