OctoPrint/OctoPrint #1097: [request] Assign different configs to different users

yoolsf, 2015-10-28:
Is it possible to assign different configs to different users?

foosel, 2015-10-28:
Please be more specific. What config settings are you thinking about? 

If you are thinking about "separate `config.yaml` files per OctoPrint user" - no, that is not possible, and it wouldn't make sense either given the fact that the `config.yaml` file is the central _server_ configuration and settings therein apply mostly to global settings that wouldn't make sense at all on a per-user base.

There is also the user settings where stuff like the user's preferred interface language are stored and where plugins can also attach themselves to store further data (e.g. something like per-user custom controls could be  provided by a plugin this way). 

So -- please motivate your request a bit.

foosel, 2015-11-05:
No feedback from OP, closing.

closed 2015-11-05 by foosel