OctoPrint/OctoPrint #1105: New and Need Help

LeonthePeon, 2015-11-02:
I've had the most frustrating time getting my RasPi/Octopi set up - and just short of wiping it clean and trying again from scratch - I thought I'd reach out.

I'm trying to link a RasPi 2 with an Ultimaker 2. I got the Octopi set up and it gets an IP address - I try pinging the router and there are some long ms times and occasional dropped packets (4 lost out of 10 - but sometimes less). (possibly Problem #1)

Although I can ping the router - Octopi doesn't show up on the router with the ip address that's supposedly assigned to the Octopi (so weird... and possibly Problem #2) 

When the stars and heavens align - I can open another computer (laptop) and can log into OctoPrint.local.  (I HAVE seen the screen for web access). Because I don't have a monitor near the printer - I do this from another room without the printer hooked up. I shut the OctoPi down - bring it out to the printer, hook it up, turn it on - camera light turns on, wifi light is on, red light on Pi is on, some green flashes here and there - try web access... None... times out. (Problem #3)

Thoughts? Other than mine that includes throwing the Octopi out of a plane...?

markwal, 2015-11-02:
From what I can tell from here, all of your problems are network connectivity related.  Unfortunately, wiping it clean and trying again from scratch probably won't help.

Since this is a support request and not a bug (yet) the best place to get help is on the google group or in the google plus community: https://groups.google.com/forum/#!forum/octoprint

You need to get your network connectivity solid enough so that when you ping the router from your pi, it comes back 10 out of 10 with a reasonably low ping time.  You'll be unhappy with using it otherwise.

When you post to the Google group, please include some extra information:

What are you using for network connectivity (which brand and model of usb wifi, for example)?
Have you tested with it wired directly to the router via ethernet cable?
Are you able to ssh into the pi using its ip address?

closed 2015-11-02 by markwal

LeonthePeon, 2015-11-03:
Thanks for the quick response Mark.  I was able to get it connected with some changes to some system files.  But still getting higher latency than I would like.  I’ll check out your link though.

I was able to get the OctoPrint screen up - just haven’t pushed any files through yet.  I was just happy to even see the screen last night and went to bed.

I’ll let you know how I do.

Thanks,
Jimmy

LeonthePeon, 2015-11-04:
Just to give you an update - I successfully sent a file to my UM2 via Octoprint and it came out well.  

I haven’t tried solving the high latency (almost don’t want to rock the boat right now).

But I did have a question - as the UM2 was not officially intended to print via USB - is it better (for workflow) to print my file through OctoPrint? Or to use Octoprint to access the SD card on the UM2; upload the file to the SD; then get OctoPrint to start the print via SD card?

Thanks,
Jimmy

markwal, 2015-11-05:
I posted the Q&A here: https://groups.google.com/d/msg/octoprint/0JaE-9J2cuw/qSKU9lakDgAJ