OctoPrint/OctoPrint #1108: [Octoprint 1.2.7 on RPI] - On connect current job stops/dies

wiredsource, 2015-11-06:
The current job will block and stop on connect from web interface.

No visible way to continue the job and therefore must be discarded.

The solution could be that the job will not be interrupted on connect or a job continue feature will be implemented...?

foosel, 2015-11-06:
@wiredsource we need logs here. This sounds like an actual issue, but without everything listed under [How to file a bug report](https://github.com/foosel/OctoPrint/wiki/How-to-file-a-bug-report) nobody will be able to help you here, especially not if you mark this ticket wrongly as being a request when in fact it is an actual bug you might be experiencing.

Salandora, 2015-11-06:
@wiredsource do you mean you started a print from your printers Sd card through the LCD.
And then try to connect Octoprint to the printer?

If yes well there is nothing we could do here because most printers are resetting when a serial connection gets established.

If no please provide more informations.

foosel, 2015-11-15:
No reply from OP, closing. Feel free to request a reopen when you can provide additional information.

closed 2015-11-15 by foosel