OctoPrint/OctoPrint #1109: Hide tab control and Terminal for non admin users using OctoPrint: 1.2.7

UM2EXTENDED, 2015-11-06:
Hi,

just signed in as I am stuck with the following using Octoprint on a UM2 Extended.

I need to hide the tab Control and Terminal for non admin users but keep the camera view without control options.
[image]
[image]

How to do this?

Salandora, 2015-11-06:
Hello,

do you want your normal users have the access to these tabs restricted or only guests which are not logged in?

UM2EXTENDED, 2015-11-06:
I want to hide it for normal (non admin) logged in users to avoid damage on the printer caused by beginner users.

Not logged in users: View only
Normal user logged in: Control (keep camera) and Terminal hidden

Is it possible?

Salandora, 2015-11-06:
That's not so easy. You would have to modify the source code of octoprint to realise that.
I can assist you if you like.

UM2EXTENDED, 2015-11-06:
Ohh, this will be way out of my limits.

Maybe it's something for the next Octoprint update, as without these settings in the user management it's just not possible to work on projects with users of different skill. Or even third party users.

Currently I can't allow some users to connect to octoprint as the risk of a printer damage due to wrong usage is to high. For Example it's still possible to lower the heater bed on position 0, beside the horrible noise it can also damage the motor very easy. Nothing you will recognize on the camera.

nophead, 2015-11-06:
Why give them an account at all then? If they don't log in they can observe
only.

foosel, 2015-11-06:
Seconding the question of @nophead

foosel, 2015-11-06:
Closing this since it is now tracked as part of #1110 (feel free to continue the discussion however).

closed 2015-11-06 by foosel