OctoPrint/OctoPrint #1111: Ability to schedule M105 requests even if not printing

szymonrychu, 2015-11-07:
After setting temperature only it would be good, that temperature graph would refresh itself every several seconds.

markwal, 2015-11-07:
This is already a feature that is in the sources.  The Serial Communications tab has separate temperature intervals for when there is a temperature set than otherwise:

[image]

markwal, 2015-11-07:
It's a new feature in the development branch as of September 3, so it may as yet be unreleased.

foosel, 2015-11-08:
M105 also gets sent when not printing, even in stable. What's new in the
development brain is that you can have different timeouts.

foosel, 2015-11-15:
No reply from OP, closing. Feel free to request a reopen when you can provide additional information.

closed 2015-11-15 by foosel