OctoPrint/OctoPrint #1112: When using remember login I can't turn on/off the printer

fraternl, 2015-11-10:
I've edited the config of octoprint on my Raspberry Pi so I could turn my printer on and off using a relay. This works fine, but not always....

Sometimes it says this is "Forbidden"
I used to reboot the Raspberry Pi, but now I noticed it's best to log out and then log in again.

If I'm logged in using the "remember" check box, this doesn't work.

nophead, 2015-11-10:
Yes I frequently get Forbidden and have to log out and in again multiple
times or send the command from another computer. What does it mean? If the
command is not allowed it shouldn't be on the menu making it impossible to
issue.

Salandora, 2015-11-11:
Please provide the octoprint.log file because there might be some informations inside.
Additionally some steps to reproduce the bug. Any information here is of help.

fraternl, 2015-12-09:
Version: 1.2.8 (master branch)

2015-12-09 17:17:29,677 - octoprint.plugins.emailnotifier - INFO - Print notification emailed to jp@xxx.com
2015-12-09 17:35:23,795 - octoprint.server.api - INFO - Performing command: sudo gpio -g write 18 1

There are no entries in the log of ~/.octoprint.log when I get my failed attempt (@ 17:30)
I then logged out and logged in.
It then responded fine.

markwal, 2015-12-09:
Probably what we need is any information that your browser console gives you on a failed attempt.  Since the server doesn't seem to note.  My guess is that there is an authentication failure.

foosel, 2015-12-09:
Please provide the content of your Javascript console (where to find? See linked contribution guidelines).

Also provide steps to reproduce. You have us flying pretty much blind here. We want to help but can't if we have no idea how to reproduce.

foosel, 2016-01-25:
Still no Javascript console or any other input that would allow to reproduce or narrow down the issue. Closing.

closed 2016-01-25 by foosel