OctoPrint/OctoPrint #1114: [Request] DisplayProgress Plugin does not work

yoolsf, 2015-11-16:
The plugin does not send the M117 commands to the Ultimaker². It is showing only the message " Printing with usb ..." on the display. Do I have to change the settings or what I have to change?
I am using a raspberry pi with OctoPrint.

Thanks for your help!

foosel, 2015-11-16:
Please report any issues or requests for plugins on the bug tracker of the plugin.

foosel, 2015-11-16:
Also in that specific case I suggest to take a close look at your terminal tab. `M117` is probably sent just fine, your UM2 is just not displaying what is being received, which would be a firmware behaviour OctoPrint or any plugin can do absolutely nothing about.

closed 2015-11-16 by foosel

yoolsf, 2015-11-23:
Ok, thanks!
The terminal at OctoPrint is receiving the M117 command. 
So it is not an issue of the plugin rather a problem of the firmware i got, right?
I have installed the original firmware 15.04.2 for the UM2. Do I have to install a custom firmware? 
If so, which one I have to install? 

Thanks for your help!

foosel, 2015-11-23:
Yes, it is a problem with your printer's firmware in that case. Sorry, I can't help you with how to get that to work, best ask in the UM community or send a mail to their support on that.