OctoPrint/OctoPrint #1116: [Request] Ability to enable timelapses during a print

nspaul, 2015-11-20:
I'd like the ability to enable/configure timelapses during a print. The use case here is that for several prints, I have forgotten to enable it, and I keep forgetting to enable between prints. If given the ability to configure timelapses during a print, it doesn't have to go ahead and start capturing. It could apply the settings once the current print job is complete.

ntoff, 2015-12-26:
@foosel How hard would it be to make Octoprint look at the file name and if the gcode or STL has the word "TIMELAPSE" in it, then it automatically turns timelapse on regardless of whether or not it's configured to do so in the settings? So you could leave timelapse turned off but for those special prints you'd just rename the file and add TIMELAPSE to the end or beginning or something.

Just a thought.

mrvn, 2016-03-11:
That might work better with support for "M240: Trigger camera". If the file contains them then the pictures would be assembled into a timelaps. Filenames are problematic, esspecially when printing a file from SD card where you only have 8+3 (actullay only  6+3 with vfat on the card) chars for the name.