OctoPrint/OctoPrint #1117: [Request] New logs created when OctoPrint is re(started)

Grogyan, 2015-11-21:
This is a request will enable me to produce a bug report with OctoPrint and Marlin.

Essentially, anytime OctoPrint starts up, create a new log, instead of deleting the old one and starting again.
This helps to diagnose issues that may be present with OctoPrint or the hardware/firmware that is connected to OctoPrint, where in such a case OctoPrint needs to be restarted/rebooted.

markwal, 2015-11-21:
The logs aren't deleted, there's a log rotator that checks sizes.  If you have sufficient space the old ones are named by date/time.  They all show up on your Logs tab in settings.  They're also easy to concatenate together using "cat" if that's what you need.

foosel, 2015-11-21:
What @markwal said. octoprint.log is continuous, rolls over based on day and keeps backups. serial.log is size limited, when it reaches 2MB it will start over. By default there is no backup for the latter (old log removed) but you can easily change that yourself in the [logging config](http://docs.octoprint.org/en/master/configuration/logging_yaml.html#changing-logging-handlers) already.

closed 2015-11-21 by foosel

foosel, 2015-11-21:
Example for setting backup for serial.log via logging.yaml:

``` yaml
handler:
  serialFile:
    backupCount: 1
```

Grogyan, 2015-11-21:
I was not aware that a backup was not produced automatically, and that it had to be setup manually to get appropriate logs of any issue where OctoPrint/OctoPi had to rebooted.
In any case, the logs I did have did not show a problem because there had to be overwritten when OctoPrint/OctoPi had to rebooted, therefore the ticket below shows to the best of my knowledge at the time to produce the appropriate log for the ticket. 
https://github.com/foosel/OctoPrint/issues/1118

I'll add the backup config into my OctoPrint setup, however I would suggest having this backup facility already enabled.  So that if someone else does come along with a problem, where OctoPrint/OctoPI SD image is loaded, then we can provide the logs to do so without extra fiddling with the base configuration files.

foosel, 2015-11-21:
You misunderstand. Only the serial log is not backups automatically since it should usually be switched off anyhow for performance reasons and only enabled when needed. The regular log does perform backups on each roll over. The only thing I can imagine that would cause your old logs to vanish was the pi crashing hard while the file was being rolled over. What entries do you have under settings > logs?

Grogyan, 2015-11-21:
Sorry, for that misunderstanding.
Yes, the PI crashed hard, as mentioned the ticket, the Raspberry PI 2 had to be completely powered off and restarted.
Unfortunately my OctoPrint server is at the makerspace and therefore cannot check that right now.  From what I remember the new entries were created daily.
As the PI2 is running continuously 24/7.
(From memory) On looking at the previous log entry everything there is what happened the previous day, whereas the current octoprint.log shows what was happening that day/today, and can be seen quite clearly, missing logged info where there should have been.  The logs are not very large typically a few kilobytes, so it isn't an overflow problem.

foosel, 2015-11-21:
For `octoprint.log` the rollover isn't size based anyhow (as written above), but if the system crashes before data can be written to disk, or if the system crashes right when a rollover happens, well, data gone. But that's not an issue with the current implementation of the logging, but rather with a host system crashing out of the blue (which certainly isn't normal behaviour, but let's tackle that in the other ticket).