OctoPrint/OctoPrint #1118: [Bug] OctoPrint Crashes which then crashes Marlin

Grogyan, 2015-11-21:
1. What were you doing?
Today, slicing a model with Cura onboard the RP2 whilst waiting for the hot end to heat up.
OctoPrint had been running for 2 weeks continuously, 24/7.
Last week, finished a print which in turn turned off power supply, hot end and steppers
OctoPrint had been running continuously for 1 week, 24/7.

2. What did you expect to happen?
Today, waiting on the hot end to finish heating up before starting the print.
Last week, expected motors, hot end and power supply to stay off.

3. What happened instead?
Today, OctoPrint crashed and forced hot end to heat up uncontrollably. OctoPrint then was completely unresponsive, forcing a full system reboot.
Last week,OctoPrint turned on the power supply and forced the hot end to heat up uncontrollably.  OctoPrint was completely unresponsive, forcing a full system reboot.

4. Branch & Commit or Version of OctoPrint:

OctoPrint 1.2.7
Raspberry PI 2

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

Mendel
RAMPS
Marlin 1.1 RC2

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

Google Chrome

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

Logs do not show crash, as they were downloaded as soon as OctoPrint/OctoPi was rebooted, which resulted in the old log being over written which would have contained the crash.
OctoPrintMarlinCrash http://pastebin.com/QMRbjHnh
OctoPrintMarlinCrash2 http://pastebin.com/Sj3edNib

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

None available at this time - was disabled, have now enabled and waiting on the crash to happen again.

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

N/A

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

http://imgur.com/RitqYIE

I have read the FAQ.

foosel, 2015-11-21:
It might make sense to first clear something about here about the way OctoPrint works.

It doesn't control your printer's heaters, motors or other functions directly. It only talks to the firmware by a more or less defined protocol of commands that allow to say things like "please heat up to $temperature", "please move the head +10mm on the X axis", "please shut off the heater (set temperature to 0)", "please turn on the printer's power supply". None of those commands will it ever send on its own. E.g. it will only heat up if you explicitly tell it to heat up, either via the temperature control panel, a gcode file that you stream _through_ it which contains a heatup command, or if you have added any commands like that to the GCODE scripts that are executed on printer connect/disconnect, print job start/finish etc.

Printer firmware on the other hand will receive those commands via the serial line and execute them. There is absolutely no way established in the "protocol" that allows the firmware to track if the host is still there (and no implementation that I know of tracks the presence of a serial connection itself), and apart from basically pinging the firmware with a temperature query every couple of seconds and seeing if there is still a reply, there is also no way for hosts such as OctoPrint to determine if the firmware is still a) there and b) responsive. 

That means that if the host issued a heatup command to the firmware (in case of OctoPrint because the user set a temperature manually or started a print which contained a heatup command), but then the host disconnects due to whatever reason, the firmware will heat up and sit at that temperature. Likewise, if the firmware crashes after that command and the connection to it drops, the host has no way at all to tell the firmware to please stop the heaters again.

So, the only way the two scenarios you encountered could happen by either your firmware (Marlin) or your host (OctoPrint) crashing "mid-print". 

You say that the OctoPrint was completely unresponsive. Could you specify this a bit more please: could you still click buttons and nothing happened or did the whole interface freeze, could you edit settings, did you try to just reconnect to the printer by chance, could you still ping the pi, did e.g. the webcam still work, stuff like that. Basically, please define "unresponsive" a bit more, since that can mean a lot in the context of a client-server-architecture, especially when three parties are involved (Single Page Application in the browser talking to the OctoPrint server, OctoPrint server talking to the printer firmware).

Looking at your screenshot I also wonder about the state of your firmware/printer. "Heating failed, system stopped" is a message generated within the firmware and then send to the host, which then disconnects from the printer since it .. well .. stops. This is basically one of the cases where the firmware crashes (here in a more or less controlled way). That of course means that you can't issue any commands like head movement etc anymore, but the rest of the interface is still responsive and you can e.g. reconnect to the printer, modify settings etc. So, again, I need more information here what exactly did and didn't work when the system became unresponsive. That error at least indicates that something is seriously wrong with your printer and or firmware which you should fix, and OctoPrint cannot cause or prevent that issue and its only course of action when the firmware reports that is to report "Error" and disconnect.

Grogyan, 2015-12-06:
So i've captured the serial log of the crash event, however have not had time to upload it to feed it into this ticket.
Anyway, from the serial log, something crashed, causing heater ID 0 error, followed by by a "kill() called" where no such command was called.
The Marlin firmware has temperature runaway prevention, however this also, for what ever reason is not run.
From what I have observed, and this is just a guess, the USB from the Raspberry PI 2 dropped connection.
Upon noticing the crash, had experimented with what worked and didn't work in OctoPrint.
Virtually every part was still working except for issuing commands via the Command tab.  I was able to log in via the Admin account and download the logs before completely shutting down and powering off the Raspberry PI 2. 
I will have time this evening to try again to upload the logs

foosel, 2015-12-06:
As I said, octoprint disconnects from printers which report unrecoverable errors. Naturally you cannot send commands to a printer you are not connected to. Even wide, when marlin reports that it's calling "kill()" that means it also will not accept any commands from that point onward, until you per cycle it.

So far I do not see any unexpected behaviour here. Your firmware crashes/kills itself due to heater errors. There is nothing octoprint can do about that, you need to find out why your printer is freaking out and fix that before you can think about using it reliably in any way.

foosel, 2015-12-06:
Just to clarify, "kill" is nothing that OctoPrint sends either. It's an internal function in Marlin that it calls itself when things go extra wrong.

Also, you keep saying that OctoPrint crashed. To be clear here as well, no, it doesn't. It really just disconnects from a printer that has reported a very serious error that is unrecoverable. And that error is also most likely caused by something OctoPrint did (OctoPrint cannot shoot heaters to bits).

So.. Let's have a look at that serial.log of yours, but I'd be very surprised from what you described so far if that would hint at something wrong on OctoPrint's end here. Or on Marlin's end for that matter. All in all this sounds more like a hardware issue with your printer than anything else (faulty heater/temperature probe).

foosel, 2016-01-25:
Still no `serial.log`. Closing.

closed 2016-01-25 by foosel

hroncok, 2016-04-25:
This just happened to us as well.

Content of terminal tab:

```
Send: G91
Recv: ok
Send: G28 X0 Y0
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G28 Z0
Recv: ok
Send: G90
Recv: ok
Send: M105
Recv: ok T:206.3 /0.0 B:83.4 /0.0 T0:206.3 /0.0 @:0 B@:0
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 G90*17
Recv: ok
Send: N2 M83*26
Recv: ok
Send: N3 G28*16
Recv: ok
Send: N4 G1 Z10*87
Recv: ok
Send: N5 M190 S80*85
Recv: ok
Send: N6 G1 Z0.4*126
Recv: ok
Send: N7 G92 E0*64
Recv: ok
Send: N8 M109 S230*97
Recv: T:197.7 E:0 W:?
Recv: T:196.8 E:0 W:?
Recv: T:195.9 E:0 W:?
Recv: T:195.4 E:0 W:?
Recv: T:195.0 E:0 W:?
Recv: T:195.0 E:0 W:?
Recv: T:195.0 E:0 W:?
Recv: T:195.0 E:0 W:?
Recv: T:195.4 E:0 W:?
Recv: T:195.6 E:0 W:?
Recv: T:195.9 E:0 W:?
Recv: T:196.4 E:0 W:?
Recv: T:197.3 E:0 W:?
Recv: T:197.7 E:0 W:?
Recv: T:198.2 E:0 W:?
Recv: T:199.1 E:0 W:?
Recv: Error:Heating failed, system stopped! Heater_ID: 0
Changing monitoring state from 'Printing' to 'Error: Heating failed, system stopped! Heater_ID: 0
'
Recv: Error:Printer halted. kill() called!
Changing monitoring state from 'Error: Heating failed, system stopped! Heater_ID: 0
' to 'Closed'
```

foosel, 2016-04-25:
```
Recv: T:199.1 E:0 W:?
Recv: Error:Heating failed, system stopped! Heater_ID: 0
Changing monitoring state from 'Printing' to 'Error: Heating failed, system stopped! Heater_ID: 0
'
Recv: Error:Printer halted. kill() called!
Changing monitoring state from 'Error: Heating failed, system stopped! Heater_ID: 0
' to 'Closed'
```

Your firmware killed the whole printer communication. That's nothing caused in OctoPrint, something is seriously wrong either with your printer's heater or your firmware. Check all your connections, thermistor, your firmware config etc. If your firmware decides to do that emergency shutdown, there's nothing *any* host software could do to make things work again.

hroncok, 2016-04-25:
I think the same thing, just wanted to post the log here.
Related MarlinFirmware/Marlin/issues/2677

Grogyan, 2016-04-25:
I have done extensive testing, and I agree with @foosel it is not
OctoPrint, it is the firmware.
Marlin RC6 should have fixed this