OctoPrint/OctoPrint #1121: Timelapse is cancelled when imagegrab fails

amedranogil, 2015-11-26:
While testing the timelapse feature (on Z change, 25fps, 10 sec post roll) with a network camera (AXIS 2130 PTZ) the timelapse procedure was cancelled after one image failed to load.

The camera I use is really old, and buggy, sometimes it gets into a state where it does not serve the image. In these cases a camera restart does the trick. After starting the print with timelapse, the camera failed (on the 3rd layer), I managed to detect it and restart the camera before the next snapshot was due. The next snapshot never took place. In fact it seems the whole timelapse procedure was abandoned (the print continued correctly fortunately). That is except for the post roll, where in fact the corresponding images where correctly grabbed. Even though the collected images where located in the timelapse/tmp folder, no movie file was generated.

It is expected that some doggy cameras can fail to provide the image, and even not so doggy (e.g: due to power fail, network fail, etc..); this does not mean the timelapse should end. I would have liked to have the final timelapse, even if one or several frames where droped.

foosel, 2016-01-25:
I tried to reproduce this by starting a print with timelapse, then disabling the webcam during the print for a couple of seconds, then re-enabling it. The timelapse capturing picked up again, an error about the capture failing was logged to `octoprint.log` (I'm missing that one from the one you linked above) and also the timelapse rendered despite the missing images.

So I really can't reproduce this. Any more hints?

amedranogil, 2016-01-25:
I really can't explain, it seamed like the image grab caused a fatal error that pushed the hole task to stand still (I'm assuming the post roll is launched as another task).
Maybe it has to do with the common error of ffmpeg when dealing with zero length or corrupt files. Maybe it is more related to the image not loading correctly than a timeout. So in this case it has to do with the camera it self (when it fails loading an image takes for ever, not timing out; I really don't know what is going on).

In any case, I haven't had problems with it, and I am satisfied that your tests are successful.

closed 2016-01-25 by amedranogil