OctoPrint/OctoPrint #1122: Toolhead Offset M218 command ignored (mostly)

MartinMajewski, 2015-11-27:
Hi,

I am using Simplify3D for a while now and have noticed a strange behavior in OctoPrint: Having a DualExtruder machine (MendelMax 3) my primary extruder has an offset that is represented in gcode via the M218 command. The MendelMax3 firmware supports this command and sometimes the exact same gcode files are positioned correctly on the bad. Bust most of the time they are not and this is not an issue with the firmware but with OctoPrint, because the prints are also wrongly positioned at the GCode Viewer when the problem happens. Somehow it seems that OctoPrint is ignoring this command or maybe it calculates the offset against its own printer settings the wrong way.

The only workaround is to apply the offset to the GCode coordinates, but then I get a wrong visualization in Simplify3D.

I also use Slic3r a lot and had never this issue with it because Slic3r does not use M218 as it seems - but since Simplify3D has a better support generator it is currently my preference.

I attached two screenshots of my printer settings and the GCode Viewer.

Please look into this issue because it drives me nuts.

[image]
[image]

Have a nice day,

Martin

foosel, 2015-11-27:
Please provide an example GCODE file with which to test.

nophead, 2015-11-27:
>because the prints are _also_ wrongly positioned at the GCode Viewer

Octoprint doesn't influence where the object is printed on the bed so surely the gcode is wrong. It might not display it correctly in the gcode viewer, but if it corresponds to where it prints on the bed then it seems it is correct.

The slicer and firmware determine where the object is printed. OcotPrint interprets the gcode to show the preview but it doesn't modify what gets send to the printer.

foosel, 2015-11-27:
@nophead good catch, I read the request of the OP as only refering to the GCODE viewer (which actually would need adjustment to be able to handle M218).

@MartinMajewski  All that can be done in OctoPrint is changing the GCODE viewer to support M218 (as stated in the updated ticket title). For the rest you'll need to approach your firmware supplier.

MartinMajewski, 2015-11-27:
https://dl.dropboxusercontent.com/u/4928832/gear_outer_f.gcode