OctoPrint/OctoPrint #1123: Temperature targets changing 'randomly'

domwakeling, 2015-11-27:
Have an issue with temperature control; whenever I start a print Octoprint raises the print-head temp to 215C (my set target) but then immediately drops the target back down to 150C.

I've tried manually changing the temperature back to 215C at this point - which worked - but it then randomly reduces the target temperature again after a few minutes.

Experiencing the same issue with two separate installs, one on a RPi using OctoPi and the other on a Mac. Have also tried using pre-sliced gcode and using the CuraEngine plugin and get the same issue in both cases.

foosel, 2015-11-27:
First of all please [read this FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-is-telling-my-printer-to-do-wrong-things). 

If you are still convinced that this is an OctoPrint bug afterwards please provide more information as requested by the bot above. Make sure to also include a `serial.log` please. You can find information on how to get that in [How to file a bug report](https://github.com/foosel/OctoPrint/wiki/How-to-file-a-bug-report).

domwakeling, 2015-11-27:
Missed that in the FAQ

The temperature drops are certainly being sent from OctoPrint, so I'll have a look at the gcode and see whether it's triggered in there.

foosel, 2015-11-27:
> Missed that in the FAQ

Not your fault, just added it this morning (it's a topic that keeps coming up)

domwakeling, 2015-11-27:
Nope, not in the GCODE - there's exactly two temperature calls in there, one to set it to 215C and one to turn the heater off at completion. I'm using the "M3D-FIO" plugin to let OctoPrint talk to an M3D Micro - sounds like that's the next suspect?

foosel, 2015-11-27:
Yes. I'd normally say, try without the plugin, but I guess that's out of the question considering the printer in question.

What you could do though is enable the virtual printer (paste ``[["set", "devel.virtualPrinter.enabled", true]]`` into the [YamlPatcher Plugin](http://plugins.octoprint.org/plugins/yamlpatcher/), save, restart and reload page) and "print" with that, see if you see the same behaviour there.

domwakeling, 2015-11-27:
OK, will try that tonight and see how it gets on. Thanks.

domwakeling, 2015-11-27:
Right, not sure quite how that's happened but problem appears tentatively to be solved; the temperature controls went nuts with the virtual printer and then started to behave themselves; re-enabled M3D-FIO and test print has held temperature correctly.

closed 2015-11-27 by domwakeling

domwakeling, 2015-12-11:
Turns out that M3D-Fio only uses the temperature setting from the M3D-Fio tab ... simple when you know how