OctoPrint/OctoPrint #1125: Printer profiles that don't have a Heated Bed should never send bed temperatures or setpoints

pyjamasam, 2015-11-29:
When a printer profile is setup and doesn't have a heated bed there should never be any data sent to the client pertaining to the bed.

I run smothieware, and in the process of calibrating my delta I guess octoprint found something in the serial stream that it thought was a temperature line that had a bed temperature in it.

*Marlin exposes this problem much more since its standard response to a M105 includes a bed temperature all the time (even if it's 0)*

After that point whenever I load the UI there is a bed temperature displayed (as well as a control to set the bed temperature - which won't actually work because I don't really have a bed).

If I disconnect and reconnect to the printer then the bed ui related components no longer appear (until the serial stream parser finds something that it things is a bed temperature again)

I understand that fixing this issue really just masks the mistake the parser is making in finding the bed temperature, but that parsing code is already making a large number of assumptions and I think it would just be easier to clean up the display and UI code.

While I realize that this isn't really a show stopper, I just find it to be something of a polish point.

Though I would pop an issue for it.

If you need any more info I'd be glad to provide what I can.

chris.

foosel, 2015-11-30:
Should be solved in `devel` and `maintenance` (and hence will be part of 1.2.8 release).

If currently selected printer profile doesn't have a heated bed, not only will no UI elements for bed related stuff be displayed (including the temperature curve), the REST API will also return a `409` if clients attempt to send bed commands, the printer state resource will have the bed temperature data stripped, no bed temperature will be reset to 0 on cancel and `M112` and also the comm layer in general will refuse to send `M140` or `M190` and instead log a warning to the terminal/`serial.log` why it's not sending those commands.

closed 2015-11-30 by foosel

voglster, 2015-12-04:
As a note, for people upgrading dev as it comes out.  My heated bed disappeared when this patch was pushed.  For anyone who runs into this just go to your printer profile and checkoff heated bed.

Before you do your next release foosel you might want to comment on this or turn on the heated bed by default so behavior doesn't change.

foosel, 2015-12-05:
It will be mentioned in the release notes as always. In fact, it is already part of the change log on the maintenance branch. Might add an additional heads up. But I guess people will still stumble over that anyhow, considering the bad track record of people reading documentation I provide ;) 

Making heated beds across all printer profiles a default however is imho the wrong approach, that would mask a configuration error on the user's part and might lead to larger problems in the future.