OctoPrint/OctoPrint #1126: [Request] M300 payload

MoonshineSG, 2015-11-29:
Can the event "Alert" get the parameters as payload (let's say as "sound") ? 

I'm trying to make the RPi play sounds, MP3, MIDI...  instead of the printer itself. Like this I could send a `M300 whateveriwant` and a script (or a octopi plugin?) would play the respective sound. Marlin would ignore other parameters other than 'P' and 'S' 

To test my scripts I added 

```
				if gcode and gcode in gcodeToEvent:
					# if this is a gcode bound to an event, trigger that now
					if gcode == "M300":
						payload = dict(
							sound=cmd[4:].strip()
						)
						eventManager().fire(gcodeToEvent[gcode], payload)
					else:
						eventManager().fire(gcodeToEvent[gcode])

```
in comm.py in `_sendCommand` but I am sure there is a better solution...

MoonshineSG, 2015-12-01:
Never mind this one.. I am writing it as a plugin. But I have a question... Is there a way for a plugin that implements "octoprint.comm.protocol.gcode.sending"  hook to write a short message to the web based terminal ?

markwal, 2015-12-01:
@MoonshineSG You might be interested in this plugin: https://github.com/AmedeeBulle/StatusLine

MoonshineSG, 2015-12-01:
@markwal No, I want to write on the Terminal window.

foosel, 2015-12-01:
It's currently undocumented, but the comm object (which the hook should provide you with) does have a method `_log` with which you can write to the terminal. That's what you want to use here.

MoonshineSG, 2015-12-01:
Doesn't seams like comm is available ...Maybe indirect ?

markwal, 2015-12-01:
First parameter to gcode.sending is comm. http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase

foosel, 2015-12-01:
[See the docs](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase) - very first parameter to the hook.

MoonshineSG, 2015-12-01:
yup.. it's there! :D and it work perfectly! Thank you!

foosel, 2015-12-01:
Can we close this then?

MoonshineSG, 2015-12-01:
Yes, since this in no longer a valid request.

closed 2015-12-01 by MoonshineSG