OctoPrint/OctoPrint #1135: Update to 1.2.8 doesn't work

Wanegain, 2015-12-09:
Hi all !

I am using a rpi2 with Octopi with 1.2.7 version.
The system tell me an update is available for OctoPrint (1.2.8). I click on proceed and a message appears : Update successful!
But I'm still in 1.2.7 and again the system tell me an update is available...
I tried to restart OctoPrint and to reboot the rpi but stil the same issue...

Thank you for investigation !

foosel, 2015-12-09:
Please read the [FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#im-running-127-i-tried-to-update-to-a-newer-version-via-the-software-update-plugin-but-im-still-on-127-after-restart), the [release notes](https://github.com/foosel/OctoPrint/releases/tag/1.2.8) or the [G+ release announcement](https://plus.google.com/u/0/b/110130855001142142895/+OctoprintOrg/posts/VAttLfxezFN), all of them explain why that is and how to solve it.

foosel, 2015-12-09:
Leaving this open so people can find it better.

Wanegain, 2015-12-09:
Oh... Really sorry for that ! But thank you a lot for the information.

AndKe, 2015-12-09:
maybe a link to releasenotes in the update notification would help people find such info.

foosel, 2015-12-09:
It might and that's why it's already on the TODO list. Until that is the case I suggest at least skimming the release notes on updates - they are always available at https://github.com/foosel/OctoPrint/releases/

foosel, 2016-02-15:
Closing this again. It is fixed after all and was only kept open to help users that don't know about the FAQ or the release notes to find it.

closed 2016-02-15 by foosel