OctoPrint/OctoPrint #1136: Cancelling print print gives "Rendering of timelapse" error.

AndKe, 2015-12-09:
Cancelling a print (before reached temperature, so no Z change, and pictures are taken)

gives this error a while later.

Rendering of timelapse Pitot-mount-for-Observer_front_20151209162031-failed.mpg failed with return code 1

It seems the video job is not cancelled.

markwal, 2015-12-09:
We don't want to automatically cancel creating a timelapse on a failed print.  On the contrary, it may be useful to watch the timelapse on a failed print.  At most, we could suppress the warning when it isn't surprising that there are no snapshots yet, but that would seem like more work than it would be worth (since we'd want to be careful not to suppress a useful warning when the lack of snapshots is unexpected).

foosel, 2015-12-09:
If the lack of pictures is the problem here (oh log where art thou?), a show horned solution for that corner case would be to simply take a picture the very first thing at print start instead of waiting for first interval or z change.

closed 2015-12-24 by GitIssueBot