OctoPrint/OctoPrint #1137: "Restart failed" neeeds a bit longer timeout

AndKe, 2015-12-09:
On OctoPi , after each upgrade, I see the red message about failed restart.
- 5 seconds later connection is lost due to reboot.
A little longer timeout would be nice.

foosel, 2015-12-09:
Quoting the [release notes of 1.2.8](https://github.com/foosel/OctoPrint/releases/tag/1.2.8):

> Increased timeout of "waiting for restart" after an update from 20 to 60sec (20sec turned out to be too little for OctoPi for whatever reason).

Naturally, that didn't apply to updating to 1.2.8, but should from now on.

closed 2015-12-09 by foosel

AndKe, 2015-12-09:
Thanks. The selecting "reboot" from top menu, it looses connection surprisingly fast.
I guess that was due to updates/changes that are still being writen, maybe slow SD ?  - *or* the fact that the system is always on, and after an upgrade it have been on for a month or more, but I don't see other performance degradation  - so never thought of it besides when upgrading.