OctoPrint/OctoPrint #1139: Canceling print does not return control to printer

foulowl, 2015-12-09:
#### What were you doing?

Procedure is as follows:
1. Click print button for any gcode file.
2. Wait for a bit.
3. Click cancel before print has started. (ie, I forgot the clear the bed, bed temp doesn't get hot enough because the hot end is too close to the bed and sucks some of the heat away, etc)
4. Control does not return to the user and the bed temp target stays the same. I can't move the extruder or execute commands in the terminal.

#### What did you expect to happen?

I would expect control to return to the user. 

#### What happened instead?

It seems to be still trying to print/won't return control to the user.

#### Branch & Commit or Version of OctoPrint

1.2.7 master and 1.2.8 master.

#### Printer model & used firmware incl. version

Prusa i3 & Marlin 1.0.2 

#### Browser and Version of Browser, Operating System running Browser

Iceweasel 38.3.0, Debian 8.

#### Link to octoprint.log

https://gist.github.com/foulowl/7a72b4a1f4713d26ff35

#### Link to contents of terminal tab or serial.log

No content in serial.log.

#### Link to contents of Javascript console in the browser

No content.

#### Screenshot(s) showing the problem:

http://i.imgur.com/oloqfSD.png

Note this is after clicking cancel and waiting a few minutes. 

I have read the FAQ.

markwal, 2015-12-09:
This is somewhat "by design".  Cancel means stop sending commands to the printer and when it sends "ok" meaning it is ready for some serial communication, send the user configurable "cancel" gcode script.

For many mainstream firmwares, this is designed into the firmware.  The only "interrupt" wait, long moves or jump the internal command buffer is "M112" which only made it into the Marlin sources last year.  M112 is defined as "emergency shutdown" so it does do the interrupt (if you have a recent firmware), but in Marlin at least, it goes into a halt loop and the printer has to be reset before you can continue.  So having the cancel print button send M112 is problematic.

You could add a custom control to the "Control" tab that sends the M112 if you like.  And I believe we have it wired up now so it won't wait for the ready state to send it.  You could also have a plugin that watches for the PrintCancelled event and sends the M112 automatically (it does this for MakerBot/FlashForge using the GPX plugin), but whether this works well will depend on what your firmware does with M112.

Firmwares really ought to have an "abort = cancel all pending operations, shutoff all heaters and motors" that is different than "emergency shutdown".

foosel, 2015-12-09:
Too clarify, a cancel should stop sending data from the print job to the printer but still accept commands.

But if the firmware is stuck waiting for a target temperature to be reached that will never be reached, it doesn't accept commands anymore, and this is probably what is the cause here. There is nothing OctoPrint can do about that since the firmware simply is not responding anymore (only printing the current vs target temperature to serial).

@foulowl To be fully sure that is the cause here, please provide a serial.log (the contribution guidelines explain how to enable it, it's disabled by default which is why it was empty ibn your case), or alternatively the connects of the terminal tab. Considering the description of the scenario where it occurred though, it definitely sounds like the usual "firmware is stuck in an endless heat loop" issue, and that needs to be solved firmware side.

foulowl, 2015-12-10:
Ah yes indeed, it looks like the printer is waiting for a temp that will not get reached. 

A way that Octoprint could handle this case (not saying it should necessarily) is to do the workaround which I have been using, either reboot or disconnect/reconnect.

Let me enable serial.log and confirm.

Thanks!!

foosel, 2015-12-10:
Maybe someone will write a plugin to do that, but I'd be all against doing something like that in the core software - trying to be smart about things like that usually leads to more trouble ("My printer suddenly disconnected just because my heater took a bit longer to heat up!") than it's worth ;)

markwal, 2015-12-10:
@foulowl, does your printer handle M112 well? You could type it into the terminal window when you want to cancel. I could make a plugin that sends m112 on cancel.

ntoff, 2015-12-29:
@foulowl My sneaky way around this is to cancel and then go into connection and hit disconnect and then reconnect. Typically this will send a reset command to the arduino MCU and the printer control will return to the user.

Results may vary if you aren't using an arduino mega + RAMPS or equivalent.

jminardi, 2016-01-22:
`M410` in Marlin cancels all moves (event those currently executing) but does not require a disconnect and reconnect. I use this to get faster cancels. Here is my "After print job is canceled" GCODE script:

```GCode
M410
G4 S2
G91
G1 Z20 F500

G28 X Y

;disable all heaters
{% snippet 'disable_hotends' %}
M140 S0

;disable fan
M106 S0
```

For some reason I found that 2 second pause is needed for the rest of the script to work.

foosel, 2016-01-25:
@foulowl any update on this?

foosel, 2016-02-15:
Apparently not, closing then.

closed 2016-02-15 by foosel

adrianmihalko, 2018-04-29:
Today I canceled a print, but Octoprint stuck in a "Canceling state". After 15 minute I restarted Octoprint.

[image]

mrvanes, 2018-05-06:
I came here because I also experience "stuck in canceling" octoprint behaviour.
Octoprint: 1.3.8. I was thinking my No Temp terminal filter was a bit too eager and maybe therefor octoprint doesn't see the ok's (just guessing here)? Anyway, I've adjusted it to not eat the "ok T:"s and I'll let know what happens.
For now, only restarting Octoprint helped:
These are the lines following the cancel command:
```
May  6 21:23:13 octoprint octoprint[869]: 2018-05-06 21:23:13,599 - octoprint.server.heartbeat - INFO - Server heartbeat <3
May  6 21:28:37 octoprint octoprint[869]: 2018-05-06 21:28:37,721 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
May  6 21:31:02 octoprint octoprint[869]: 2018-05-06 21:31:02,333 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
May  6 21:31:02 octoprint octoprint[869]: 2018-05-06 21:31:02,338 - octoprint.server.util.sockjs - WARNING - Could not send message to client 127.0.0.1: object of type 'NoneType' has no len()
May  6 21:31:03 octoprint octoprint[869]: 2018-05-06 21:31:03,722 - octoprint.server.util.sockjs - INFO - New connection from client: 127.0.0.1
May  6 21:31:09 octoprint octoprint[869]: 2018-05-06 21:31:09,666 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 0.56s
```
(I was working on the machine, hence the localhost connection)

foosel, 2018-05-08:
Anyone experiencing being stuck in canceling or pausing on 1.3.7/1.3.8 should take a look at [this FAQ entry on that matter](https://discourse.octoprint.org/t/octoprint-1-3-7-1-3-8-is-stuck-in-pausing-or-cancelling/1625) or at least the relevant ticket #2581 that also contain a workaround instead of this ancient ticket from a time when the Pausing and Cancelling state didn't even exist yet.