OctoPrint/OctoPrint #1140: [Bug] Printing from LCD UltiController SD Card Fails to start

Brictone, 2015-12-10:
#### What were you doing?
Trying to print from Gcode on SD Card in the LCD UltiController through the Octoprint web interface.

#### What did you expect to happen?
The Printer to begin printing Gcode file until completed

#### What happened instead?
Octoprint reports Calculating...  indefinitely
The Ultimaker controller just reports the file name in the truncated form 12toot ->1.gco and the print duration.  

Also Another note on this and I believe it may be related.  If I have the pi plugged into the printer with the USB cable and do not connect from within the software. While Auto connect is off, if I unplug the USB, the print continues fine until I plug the USB back in and the printer halts in it's tracks and Octoprint remains unconnected to the printer. (I believe this is an improvement from 1.2.7 where it would halt if the pi was disconnected)

Another thing is the SD Card files are truncated in Octoprint to 8 Characters and 3 extension characters like in the old Days.  The LCD Controller doesn't do this.

These bugs might fall under separate bugs but they are all related to LCD Controller printing while connected to Octoprint

#### Branch & Commit or Version of OctoPrint
1.2.8 (master branch)

#### Printer model & used firmware incl. version
https://ultimaker.com/en/community/8807-main-links-for-marlin-firmware
https://github.com/..._UM_HeatedBedUpgrade

#### Browser and Version of Browser, Operating System running Browser
Google Chrome

#### Link to octoprint.log
http://pastebin.com/NdMHG0Xm

#### Link to contents of terminal tab or serial.log
http://pastebin.com/3CELxBbE

#### Link to contents of Javascript console in the browser
I don't know where the javascript console is

#### Screenshot(s) showing the problem:
Machine State: Printing from SD
File: 12toot~1.gco (SD)
Timelapse: Timed (60 sec)
Approx. Total Print Time: -
Print Time: 00:12:25
Print Time Left: Calculating...
Printed: - / 26.8MB

Truncated files from LCD Controller SD Card
12toot~1.gco
Uploaded: -
Size: -
    
1x_ell~1.gco
Uploaded: -
Size: -
    
1x_of2~1.gco
Uploaded: -
Size: -
    
201401~1.gco
Uploaded: -
Size: -
    
201402~1.gco

foosel, 2015-12-10:
You forgot the final line of the template, hence the bot.

Anyway, it looks like you should take a look at [this entry in the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#i-cannot-select-files-on-the-printers-sd-card) (actual [this](https://github.com/foosel/OctoPrint/wiki/FAQ#i-cannot-select-and-start-printing-files-on-the-printers-sd-card), but that just links to the aforementioned) and urge Ultimaker to fix their firmware (or maybe there is a fixed build already you need to flash.

Brictone, 2015-12-10:
Ugh, That's frustrating, I've been reading over all of this Octoprint stuff for the last 6 months every once in a while and I missed that part about Marlin.  I'll see If I can hassle them about fixing it for octoprint.  Thanks.

foosel, 2015-12-10:
If push comes to shove, you could always compile your own firmware. It's not as hard as it sounds :)

foosel, 2015-12-10:
Closing this since its a firmware related problem.

closed 2015-12-10 by foosel