OctoPrint/OctoPrint #1142: Bulk File Delete

sniglet, 2015-12-13:
Once the printer is up and running well, the files section fills up with prints I'll most likely never use again. A 'bulk delete' button or dialog would be nice so save people from having to SSH into the server to clean things up.

Salandora, 2015-12-13:
If you switch to the devel branch you could install my filemanager plugin, wich supports bulk remove.
Additionally you will have folder support, wich makes organizing gcode files much easier.

sniglet, 2015-12-13:
Cool. Will do (developer branch)

ignaworn, 2015-12-13:
Salandora, are you refering to this plugin? https://github.com/Salandora/OctoPrint-FileManager
I just installed and it does not seem to do anything at all. Maybe I'm not looking into the right place. 

Version: 1.3.0.dev761+g28eb436 (devel branch)

Salandora, 2015-12-13:
@ignaworn yes that is the plugin. You should have a new tab (beside Temperature, Control and the others).
Did you restarted Octoprinz after installing?

ignaworn, 2015-12-13:
Yes, I restarted OctoPrint, and RPi. Such tab never appeared.
[image]

[image]

I've also disabled all the other plugins and tested, still not wokring.
Thanks for your help!

BTW: Should we move this conversation to another place?

Salandora, 2015-12-13:
Yeah please. 
Either join irc or open up an issue on the filemanager plugin page.
No matter what please add the octoprint.log file.

eykrevooh, 2016-03-20:
The plugin mentioned above addresses this issue. Is this now a feature request for the core or should this ticket be closed?

foosel, 2016-03-21:
Might make sense to bundle @Salandora 's plugin in a future version, to have this functionality part of the standard distribution. But yes, considering that the plugin exists I don't see a reason to add this to the core itself. Leaving this ticket open though since the functionality that plugin depends on is only available on the `devel` branch for now, so it's not fully solved. Will be once 1.3.0 gets released though.

Adding this to the milestone, since the availability of the mentioned plugin to the general public depends on 1.3.0 functionality.

foosel, 2016-12-12:
1.3.0 is released, @Salandora 's plugin is available on the plugin manager for everyone to use. Closing.

closed 2016-12-12 by foosel