OctoPrint/OctoPrint #1145: [request] configurable file-is-to-large-for-gcode-preview size

mrvn, 2015-12-15:
For large files the GCode preview shows a warning that the file is too large. But what is too large? The size should be configurable so it shows up earlier on small systems.

foosel, 2015-12-15:
Currently not available via the UI and I just realized also undocumented, but can configured via [config.yaml](http://docs.octoprint.org/en/master/configuration/config_yaml.html) by setting `gcodeViewer.sizeThreshold` and/or `gcodeViewer.mobileSizeThreshold`. The expected value is the maximum file size in bytes for which to surpress the warning. Default values are 20MB for `sizeThreshold` and 2MB for `mobileSizeThreshold`. To change to e.g. 100MB (100 * 1024 * 1024 = 104857600) and 10MB (10 * 1024 * 1024 = 10485760) respectively, use this YAML excerpt (e.g. via [YamlPatcher](http://plugins.octoprint.org/plugins/yamlpatcher/)):

``` yaml
gcodeViewer:
  sizeThreshold: 104857600
  mobileSizeThreshold: 10485760
```

Leaving this open as a reminder to add to the settings dialog and to document.

philphilphil, 2016-04-05:
I would grab that i it's ok, seems like a good one to start

foosel, 2016-04-05:
@philphilphil considering it grabbed by you then!

foosel, 2017-01-27:
Should have been closed back in April since it's implemented. Rectifying that now ;)

closed 2017-01-27 by foosel