OctoPrint/OctoPrint #1146: G1 E25 F300 is not applied in the start qcode

Shantos, 2015-12-15:
Hi,
I will let extrude my printer 25mm filament at the beginning of the printing process. But if I use octoprint the code G1 E25 F300 is only sometimes used. It works every time if I print directly from my computer.  

I´m using Simplify3D, and the start code is in the gcode.

This is my Starting Code:
G28 ; home all axes
G1 E25 F300 
G1 Y10.0000 F6000

The code befor and after the  G1 E25 F300 is working fine.

Did you have any idea what I'm doing wrong?

thx
Shantos

markwal, 2015-12-15:
Are you saying that the G1 E25 F300 isn't sent to the printer?  If so, please create and attach serial.log (you have to turn it on first since it is big and takes resources).

Or are you saying it appears to have a different effect when it is sent (it still shows up in the terminal window) it just doesn't do what you expect.  If you are using a GPX plugin, this might be because it is mistaken about the extruder starting position which I can help you with.  Please attach an octoprint.log.

ntoff, 2015-12-29:
always always ALWAYS zero the extruder before an extrude move because they have no home switch and no way of knowing where zero actually is.

G92 E0 in your start code before and after your (I assume) prime extrude should help.

closed 2015-12-30 by GitIssueBot