OctoPrint/OctoPrint #1150: [BRAINSTORMING] Speed initial load time with asset merging & compression

flyswatter, 2015-12-19:
Hi there, web developer here, I love my OctoPrint!

I noticed the initial page load time is around 4.5s on my Pi B+, and some brief investigation revealed the sheer number of assets being requested as a probable source of slowdown:

[image]

The green areas are being spent waiting for responses, and the blue areas are spent actually transferring information.  If we only had the green wait once, and loaded all the information at once, I think we could speed up the process a lot!

I've only used Jinja for a couple projects before, and I'm not big with Flask (or even Python these days), so I figured I'd be better off starting a discussion than diving into an under informed solution.

It looks like the python library [webassets](http://webassets.readthedocs.org/en/latest/index.html) easily does what I'm describing, and is compatible with the current stack.

Thoughts?

flyswatter, 2015-12-19:
I didn't even mention minification or compression in the original post, but both of those would be additional performance benefits of an asset compiler like `webassets`.

foosel, 2015-12-19:
Which version were you looking at? Asking because 1.2.x in fact uses webassets and bundles css, js and less files. At least if you didn't disable that via a devel config flag. Minification can in principle also be enable via a config flag but I noticed issues with that with a couple of libraries and most importantly don't want to risk killing the full app if a plugin with sloppy js is loaded which when minified produces invalid js.

Btw, just bundling he assets didn't yet speed up things enough, so in 1.2.9 there will also be some more intelligent server side caching plus more intelligently calculated caching headers to allow the browser to cache more aggressively too.

flyswatter, 2015-12-19:
Awesome update, thanks!

I recently pulled the latest from master and ran the install script and got plugins loading, (hadn't updated in maybe a year) but it only says 1.1.something, I must have installed wrong.

Any tips on that? I suppose we can close this.  Sad to hear it didn't help much, but awesome that you're on to the next optimizations!

- Dan

foosel, 2015-12-19:
Sounds like you didn't properly install the update. Current master is 1.2.8. In general: make sure you run `python setup.py install` with the correct python instance - on OctoPi this must be `~/oprint/bin/python`, not just `python`. Also no `sudo` anywhere. See also [the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-update-the-octoprint-installation-on-my-octopi-image).

And to correct myself there, it *did* improve things, but with the new plugin system in place in 1.2.x, fires started burning elsewhere (rendering the UI HTML itself), so some more tweaking was necessary.

Anyhow, closing this :)

closed 2015-12-19 by foosel