OctoPrint/OctoPrint #1153: [Request] System command type in Controls structure

naterade, 2015-12-21:
Related to closed case #1095. I believe it's worthy of another look.

Currently, to create custom system commands a user must use the Event or System structures of config.yaml.

In my specific case, I would like to send a parameter to the system for use with RPi-gpio applications. This would be most intuitively accomplished through the controls structure with a slider or value-entry field. 

In the Events structure, the command type is configurable between system and printer/g-code types. 
Rather than use a Controls command to send a "dummy" printer command (M300 or other) to trigger an event to send a system command, adding the "type" switch to the Controls structure would allow the script to be sent directly from the Controls command without limiting the user to the predefined Event list or the static nature of the Systems tab.

controls:
..-.children:
....name: xyz
....width: '2'
....type: system <-------------------------------------- new option
....command: gpio abcParameter
....input:
....-.default: 10
......name: abcParameter
......parameter: abcParameter

jneilliii, 2022-01-23:
gcode system commands allows for passing environment variables to system commands, which could then be integrated as a custom control with input?

CmdrCody51, 2023-03-30:
You mean like this?
[image]
The first two are filled with defaults and the Value is blank.
<
  - command: OCTO400 File=%(myf)s Name=%(myn)s Value=%(myv)s
    confirm: You are about to modify the run time parameters of your Macro files!
    input:
    - default: WhyMe.data
      name: File
      parameter: myf
      slider: false
    - default: mine.my_
      name: Name
      parameter: myn
      slider: false
    - default: ''
      name: Value
      parameter: myv
      slider: false
    name: Parameters
>
With GCODE System Vommands OCTO400 defined as:
/home/pi/.octoprint/scripts/Variable.sh
That is passed dqta as:
G-Code arguments will be passed to executed system commands via environment variables. The following will be set:
OCTOPRINT_GCODESYSTEMCOMMAND_ID
OCTOPRINT_GCODESYSTEMCOMMAND_ARGS
OCTOPRINT_GCODESYSTEMCOMMAND_LINE

One thing I like about this is the fact that that entry format is usable **from** GCode as is!
So in my startup I can use :
OCTO400 File=WhyMe.dat Name=mine.my_FirstLayerChk Value=true
and OctoPrint passes that data  to the script.

I think this even more than solves this issue.