OctoPrint/OctoPrint #1154: Fresh install: firstRunSetup addUser AttributeError: 'NoneType' object has no attribute 'addUser'

emcniece, 2015-12-22:
Trying to get a new setup rolling, but the app encounters a few 403 and 500 errors on the front end. Log inspection reveals the following dump after forcing firstRunSetup: true in config.yaml:

    File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.2-py2.7.egg/octoprint/server/api/__ini__.py", line 112, in firstRunSetup
        AttributeError: 'NoneType' object has no attribute 'addUser'

[Terrible photo of stack dump here](http://i.imgur.com/AwmOPAX.jpg)

Yes I saw the guidelines and yes this is on a rPi, but this appears to be related to the python server API. Has anyone hit this before? Please close if not relevant.

markwal, 2015-12-22:
There is something broken in your config.yaml or users.yaml that leads to the userManager being None.  If your copy both to a backup location and then delete them from ~/.octoprint, you won't hit this exception any more.

emcniece, 2015-12-26:
Success! Thank you for the tip.

closed 2015-12-26 by emcniece