OctoPrint/OctoPrint #1156: [BUG] Print starts right away after uploading a gcode file...

bolsoncerrado, 2015-12-24:
...if you press PRINT.
So instead of waiting for the temps to be reached and homing the head, the print starts "as is"... Makes no sense :????

nophead, 2015-12-24:
It does whatever the gcode tells it to do. If you want to home and heat you need that at the start of your gcode. Most slicers allow arbitrary commands at the beginning and default to heating and waiting.

bolsoncerrado, 2015-12-24:
The gcode is meant to have those instructions inserted. What I mean is if you click PRINT right after upload happents, the printer bypasses the BEGIN script, for whatever reason.

foosel, 2015-12-24:
Well... I suggest to check your GCODE then, because for whatever reason it doesn't have those instructions inserted. OctoPrint doesn't care whether you hit "Print" directly after upload or three years later, it will always send whatever is in the file to the printer. So the problem must be either in the file (the start GCODE not being in there as you expect it to be) or with your printer.

Other than that - @GitIssueBot told you about some things that we need here in order to be able to help *at all*. Without that information, nobody can help you. It's not the first time that you've ignored that request for info ;)

ntoff, 2015-12-26:
@bolsoncerrado are you uploading an STL and having octoprint slice it? Or are you uploading a straight gcode file?

bolsoncerrado, 2015-12-26:
I use the SAME settings for every and all .gcodes I upload to octoprint using Simplify and it only happens if I accidentally click PRINT too early after upload concludes.... If i wait seconds enough after upload and click PRINT then everything goes as expected. Its just a matter of clicking TOO EARLY right after upload process has concluded... 

It has nothing to do with the slicer settings/software, sorry.

markwal, 2015-12-26:
To debug, we will need a serial.log from when it apparently didn't send part of the gcode file to the printer and the gcode file.

ntoff, 2015-12-29:
I wonder if this might be half related to the whole "print button is enabled regardless" bug (I forget the issue number). The print button is active and able to be pressed even without a print loaded so octoprint doesn't seem to know or care whether or not the code is actually completely uploaded and loaded or even if *any* code is loaded, it'll always let you hit print.

closed 2016-01-08 by GitIssueBot