OctoPrint/OctoPrint #1164: [request] sort of idea for setting timelapse starting point

ntoff, 2015-12-29:
So my printer heats up before the print starts (duh) but that also means there might be quite a bit of video at the start of a timelapse with it just sitting doing nothing. I'm just wondering if it's possible / easy to figure out a way of starting the timelapse recording right before a print is about to start? Like maybe read the gcode and see what the extruder is set to and when it's about to reach that temp it then starts a recording? I realize not everyone would start their gcode the same (some might manually pre-heat it).

It's just an idea for discussion, not really a feature request but it would be nice.

foosel, 2016-01-11:
Just realized I forgot to reply here, sorry.

In principle I like the idea, the problem is the "analyze the GCODE part" however. That's always problematic. We don't have read access to the GCODE file if it's stored on the printer's SD card for example, but timelapses should still work then. And as you already pointed out, not everyone might be using something like `M109` for a heat-and-wait before the print, but do it differently. So the only thing that's reliable is the actual start and end of a file, everything that's in there can be anything. So stuff like that is extraordinarily tricky in the core application.

An idea might be to extend timelapses in so far that they allow better extension through plugins, in which case someone could write a plugin to wait the first n seconds or until a certain default temperature is reached (anything above 200 should mean it won't take as long), or maybe even hook into the actual heatup some way or another.

ntoff, 2016-01-12:
What then about just a simple option in the timelapse options page to start it once the hot end or heated bed reaches a certain temperature that we could manually set? No reading of gcode required and as far as I can see, if octoprint is plugged into the printer, even for an SD print it should still see the hot end's temperature wouldn't it?

ntoff, 2016-01-12:
I made a reply then realized it was exactly what you said. I like the idea of starting the timelapse once either the bed or hot end reaches a certain temperature. 

Kind of makes sense since the pictures can either be taken at timed intervals or on every layer change that there might also be an option to start the timelapse once the position reads 0,0,0 but I don't think octoprint really knows where the printer is without constantly querying it so again SD printing would fail to start the sequence but how does it work on Z change for SD printing currently? Or doesn't it?

markwal, 2016-01-12:
@ntoff timelapse is disabled if Z change and SD printing.  There's a warning to that effect in the timelapse pane (no way to detect change in Z reliably).

The typical response to the M25 query (SD card print status) includes current byte of how many bytes.  Also, M105's are still done during printing so one of those bits could be used as a start trigger though they both will have some lag due to the polling period.

There are several requests for different timelapse changes.  As foosel said, seems like one way to go would be to make more mixin/hook extensions into the current timelapse module.  Might also be a good idea to make the current timelapse module a bundled plugin that can be disabled and/or replaced by a plugin author.

foosel, 2016-02-02:
Argh, I wish there was a way to merge two tickets here... Just noticed (while looking up a bunch of other timelapse related tickets) that this is actually a duplicate of #667.

Usually I'd close this one since it came later, but since there are way more comments in this one I hope @jneilliii will be ok with closing the other one instead.

jneilliii, 2016-02-03:
no worries.  I think this is definitely something that would be helpful to implement if there is a way to do it.  Would it be possible to implement a manual start/stop of the timelapse, and expose that to plugins/gcode commands that could then be integrated into the start and end gcode of our slicers?  That way the gcode controls the start and stop of the timelapse and could be placed after bed heat up commands.

ntoff, 2016-02-04:
@jneilliii  

>integrated into the start and end gcode of our slicers?

Foosel said earlier that's not possible when you're printing from SD card because octoprint doesn't ever see the gcode.

I wonder if it's possible to put a switch on the GPIO pin of the raspi and hook it up near the end stop of the Z axis so as soon as Z gets homed octoprint could detect that on a GPIO pin and start the timelapse.