OctoPrint/OctoPrint #1168: Is this adaptable to run on a raspberry pi zero?  If so, what steps required to make it happen? TIA

datanowllc, 2015-12-31:
I have not used your software before, so I apologize if this is too simple of a question. 
I am very excited about the prospect of using your software on a raspberry pi zero.
I would like to build a Delta Smoothie 3D printer, then a larger gantry 3D printer or a multiAxis CNC machine.  
Is this adaptable to run on a raspberry pi zero?  If so, what steps required to make it happen? 
Is there a way to make a CNC mode if I give feedback to the tool head?
   Thanks in advance...

markwal, 2015-12-31:
These are good questions for the google group and/or google plus community (links are in the contribution guidlines).

closed 2015-12-31 by markwal

kolasinski, 2016-01-20:
I would like to report a bug. Octopi setup using Pi2 with tplink dongle and logiteck webcam boots, connects to network perfectly. Same card plugged into PI ZERO, but now i needed to add and IO gear otg hub, it won't boot. LED flashes 8 short one long. I researched and (on Reddit) says 8 flashes: "SDRAM not recognized." also says something about needing a newer image... when installed Octoprint image on Pi2 it updated to the latest version so I know I have the latest.... 

Ideas?  I mean,I can always just use the Pi2, but I hate to waste all that horsepower to monitor the printer... Roger  

By the way i'm on master branch 1.2.8. i didn't post to the rest of the contribution guidelines only because everything is working perfectly on the pi2, and I was hoping this is a known issue. I don't think a log is relevant (again because wont boot on the zero and works fine on the pi2. Also I tried booting only with power (no OTG hub, exactly same flashing -- of course did not expect wifi because it wasn't connected..

Please if there is something really necessary, I'm more than willing to help.

markwal, 2016-01-20:
Well, that wouldn't be a bug in OctoPrint.  OctoPrint doesn't have much to do with a Pi booting or not.  It's just a server daemon that starts up close to the end of boot anyway.  I think your problem is either with the OctoPi image not booting on a Pi Zero (last one I tried worked though).  Or with the zero itself.  So if there is a bug to report it goes either to the OctoPi project or to Raspbian.  But so far, this still looks like configuration.  (BTW, you might also try booting with the USB devices disconnected to rule out a problem with their drivers, but in this case, I think it may be your image.)

When you say "Octoprint image updated to latest version", you have to keep some separate software straight:  When reddit says LED flash 8 short one long needs a newer image, it means that it needs a newer built raspbian image (the underlying OS).  When Octoprint updates itself, it updates its own bits and any python dependencies.  It doesn't update your OS.

This is probably something you would want to post to the google groups and/or the rasberry pi forums since there may be others with your same set up that have tips for you.

jneilliii, 2016-01-20:
Looks like you have to use a nightly build of OctoPi (if it's not updated yet in official releases by now) and seems like USB 3.0 hubs may be an issue.  See post [here](http://timwappat.info/post/2015/12/06/Rasberry-Pi-Zero-OctoPi-setup) from December 6th.