OctoPrint/OctoPrint #1169: Octoprint and RPI2 GPIO

Wapata, 2016-01-01:
Hello,
I don't know if it's the best place to talk about this but:

Did someone think about a plugin that can use the RPI2 GPIO ?
I mean, let's take a moment. My printer is off, and i need to heat the extruder to take the filament pool with me. But I forget my phone (or I'm not in my wifi area). The only solution left is to cut the filament.

But can't we power up the printer and Octoprint, and have a control interface ?

Thanks for reading !

markwal, 2016-01-01:
Cut the filament is a good solution for that particular scenario. But also I have gpio connected to a powerswitch tail so I have system commands for power on off in OctoPrint.  There are guides for this in the wiki.

A good place for questions like this is the community since more people are participating there (links in the contribution guidelines linked by the bot) and this list is better used for tracking bugs.

closed 2016-01-01 by markwal

Wapata, 2016-01-01:
I see the relay solution. We can use outputs with GPIO.
And i have to look to the marlin screen control I think. Why using Octoprint if the printer itself can do it.
But hase i don't have a screen, i don't know if we can control the temperatures, I have to check.

But. Thanks for the answer ! And i will try to find the octoprint community now. And sorry for the miss-use of github.
Have a good day.