OctoPrint/OctoPrint #1170: PrintCancelled event fires as PrintFailed event type

docdawning, 2016-01-01:
My Octoprint has the version string <i>Version: 1.2.8 (master branch)</i>

My budding Octoprint "plugin" uses the event hooks to call command-line functions that cause my lighting system to change.

My <i>config.yaml</i> has the hooks as follows:
<addr>
events:
  enabled: true
  subscriptions:
  - command: echo -n "OFF;" > /dev/ttyUSB0
    enabled: true
    event: Disconnected
    type: system
  - command: echo -n "STANDBY;" > /dev/ttyUSB0
    enabled: true
    event: Connected
    type: system
  - command: echo -n "WHITE;" > /dev/ttyUSB0
    enabled: true
    event: PrintStarted
    type: system
  - command: echo -n "STANDBY;" > /dev/ttyUSB0
    enabled: true
    event: PrintDone
    type: system
  - command: echo -n "STANDBY;" > /dev/ttyUSB0
    enabled: true
    event: PrintCancelled
    type: system
  - command: echo -n "BLUE;" > /dev/ttyUSB0
    enabled: true
    event: PrintPaused
    type: system
  - command: echo -n "WHITE;" > /dev/ttyUSB0
    enabled: true
    event: PrintResumed
    type: system
  - command: echo -n "RED;" > /dev/ttyUSB0
    enabled: true
    event: PrintFailed
    type: system
</addr>

So these hooks are working well, however, I just cancelled a print and my lighting system turned <i>RED</i>. The only case where the <i>RED</i> hook is invoked is for a <i>PrintFailed</i> event. I definitely clicked the Cancel button in Octoprint, my print had only been actively running for a about 10 seconds.

Therefore, I conclude that the <i>PrintFailed</i> event has fired unintentionally.

markwal, 2016-01-02:
Actually the design is that both events fire.  A cancelled print is also a failed print.  First it fires the PrintCancelled event then the PrintFailed event leaving your LEDs red.

closed 2016-01-02 by markwal

foosel, 2016-01-02:
To elaborate on this a bit more: The idea is that if you cancel a print, it obviously failed, so both events fire. There also might be another reason for the print to fail however (e.g. a communication loss during the print), so there are two separate events. In a nutshell: if you see a `PrintCancelled` event in your plugin, ignore the `PrintFailed` event that follows.

docdawning, 2016-01-04:
Thanks foosei, works for me.