OctoPrint/OctoPrint #1171: [Request] Create an image sequence instead of video

sanchosk, 2016-01-02:
Hi.
I'd like to have image sequence instead of encoded video - I'd like the SW to keep the actual JPG files.
It would be great if the folder was named by current unix timestamp when print started instead of tmp.
I tried to find the python code for this to modify, but I'm failing.
Can someone, please, help me?

sanchosk, 2016-01-03:
Well, I managed to get together a "dirty hack" - I created a fake script and named it /usr/bin/avconv.
This is the content:
```
#!/bin/bash

# this is a fake avconv (ffmpeg) for only renaming the folder with files
# I'll try to use the file creation date to rename it appropriately
# and also rename the folder where the files are stored in

echo "Input params: $@" > /tmp/callParam

while [[ $# > 0 ]]; do
	key="$1"
	case $key in
		-i|--input)
		inputFile="$2"
		shift
		;;
		-loglevel|-vcodec|-threads|-pix_fmt|-r|-y|-b|-f|-vf)
		shift
		;;
		*)
		outputFile="${key//.mpg/}"
		;;
	esac
	shift
done

mkdir -p "${outputFile}"
[ -d "$outputFile" ] && mv "$(dirname "${inputFile}")"/*.jpg "${outputFile}/"
```
Works for me - maybe someone interested can also test it...

jneilliii, 2022-01-23:
Could you use Event Manager and the CaptureDone event to just copy the snapshot as it's been created maybe?