OctoPrint/OctoPrint #1173: Example Custom Control "Get Position" Stops Working After Several Uses

CasperH2O, 2016-01-03:
#### What were you doing?

For manual calibration of my printer (delta type) I have added some custom commands to move the head to three different locations and the "Get Position" button from the example documentation:

http://docs.octoprint.org/en/master/features/custom_controls.html#sec-features-custom-controls

Moving my print head around, and after it has stopped, checking the position with the "Get Position" button. This works for some time, but after something between 10 and 20 requests the text next to the button is no longer updated, even though I can see the correct values show up in the Terminal Tab of Octoprint. It appears to get stuck somehow?

Additionally, my config.yaml attached. [config.yaml.txt](https://github.com/foosel/OctoPrint/files/76572/config.yaml.txt)

As workarounds I have tried:
- Homing and trying again, this sometimes works.
- Refreshing the browser (F5), this does not work.
- Restarting OctoPrint, this works.

#### What did you expect to happen?

I start with my printer at its home position and move it down 1 mm. After pressing Get Position, I expect it to always show an update.

#### What happened instead?

It works about 20 times after that, the button simply shows the last position it saves. However, in the Terminal tab of Octoprint I can see the request and a correct updated position. Leading me to think something goes wrong internally in Octoprint when it receives the information and when it has to display it.

#### Branch & Commit or Version of OctoPrint

1.2.8 Master

#### Printer model & used firmware incl. version

SpiderBot 1.0 With Upgrades, Firmware: Repetier-firmware 0.91

#### Browser and Version of Browser, Operating System running Browser

Windows 7, FireFox 43.0.3
Windows 7, IE 11.0.21

Interesting to note, once the position gets stuck I can check with another browser but the button does not work.

#### Link to octoprint.log

File: [octoprint.log.txt](https://github.com/foosel/OctoPrint/files/76575/octoprint.log.txt)

#### Link to contents of terminal tab or serial.log

Terminal Tab Output: [terminal-tab-output.txt](https://github.com/foosel/OctoPrint/files/76580/terminal-tab-output.txt)

It goes wrong after location: "Position: X=0.00, Y=10.00, Z=115.00, E=0.00"

Serial Log, is empty.

#### Link to contents of Javascript console in the browser

Javascript console (firebug with FF) appears empty concerning faults and warnings.

#### Screenshot(s) showing the problem:

Screenshot of the custom control page. With it's last position update.
[image]

I have read the FAQ.

foosel, 2016-01-03:
It's a short sight in the example actually, it doesn't take potential negative coordinates into account. The regular expression simply doesn't match those, so once your X coordinates becomes negative the line isn't recognized anymore. Adjusting the regular expression to 

    X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)

should fix the problem.

CasperH2O, 2016-01-03:
Indeed is does! Thank you @foosel I have adjusted my config.yaml and it works nicely now. I can imagine the example was probably made with Cartesian printers in mind, which probably never go negative.

Update documentation and close ticket?

ntoff, 2016-01-04:
Actually my printer is cartesian and its home position is all negative (X Y and Z).

foosel, 2016-01-05:
Example in documentation fixed in 9fd8b09e34d0654df5c50470a532a935e2404eee (from which I of course forgot to reference this ticket)

closed 2016-01-05 by foosel