OctoPrint/OctoPrint #1174: [Request] GCode Viewer/Editor for Uploaded GCode

InoSiX, 2016-01-03:
Currently Octoprint has an awesome feature that shows the pattern created from uploaded GCode. I find this really cool, something that I wouldn't have expected for Octoprint running on such a low powered device. It's even fairly easy to use and understand. 

That being said, something that seems like it would be a lot simplier to implement is a code viewer/editor. I am not terribly concerned with with the print head movement per se, but things like bed heating values, extruder temp, ect.

I am currently using a FFCP, making use of the GPX plugin provided by markwal. Sometimes I export to GCode only to remember I need to change temp values, ect. This seems like a very baseline "feature" and I feel it should be shipped this way.

foosel, 2016-01-11:
This could be easily done through a plugin.

InoSiX, 2016-01-11:
The hole Octoprint could be done in plugin, with Octoprint serving as nothing else than a host for the plugins that run the whole deal. So it doesn't make sense to frame it in terms of whether or not it can be placed in a plugin, or how easy that would be to accomplish via plugin. 

I am putting forth an argument that Start and End GCode can be interacted with for small issues like temps on a daily and shouldn't need to be exported, changed, imported again to fix such issues. The GCode viewer is not a plugin, it's shipped with it. Yet this is only serves as a neat function, tracing the GCode path. The Webcam functionality is much the same. It's a neat function, but could easily be put forth as a plugin. Instead it is shipped this way.

Editing uploaded GCode is simple, and relevant directly to actually printing. It isn't a gimmick. It is relevant to all printers, not just a few. It should not be a plugin. If anything the current Viewer and webcam should be a plugin.

foosel, 2016-01-11:
Whoa, no need to get hostile, nobody said the idea was stupid so no need for that attitude.

A lot of the current core functionality will be extracted into plugins in the long run. That way users can pick the functionality they need. The GCODE viewer is a prime candidate for something like that actually, because a lot of people actually don't use it. Same would go for a GCODE editor though - not many people are comfortable enough with GCODE to actually edit it line by line, so it wouldn't make sense to force that down everyone's throat.

Plus there is maintenance (implementing something as a plugin forces decoupling and only using established interfaces, plus it allows different development cycles) and development resources (= number of people who can work on that) to consider.

The more of new functionality can be implemented as plugins (and existing functionality extracted into plugins), the better. OctoPrint acting as a platform for a lot of different "chose-your-own" functionality concerning 3d printing is actually not too far from the general vision.

The comment above was meant as a hint for potential "3rd party developers" that a GCODE editor is something that could pretty much be implemented by EVERYONE without depending on the core OctoPrint team. Actually, it would be a great exercise to get ones feet wet with the plugin system. And there's nothing preventing us from bundling commonly used plugins with OctoPrint if they gain popularity.

Salandora, 2016-01-11:
As I already said in another pos I'm working on a plugin for this as a side project I didn't had time lately wich is why the project havent progressed much.
But as foosel already said it can easily be done through a plugin.
And if you really want this feature so badly well i suggest you go and write it.

InoSiX, 2016-01-11:
That is fair, my apologies. My understanding of the plugin concept was something more along the lines of niche, specific functionality. This does not feel like niche functionality. Print temps are and things of that nature are required to get the print going, so making those types of edits seems rather baseline for more of a print-and-forget style.

foosel, 2016-01-11:
just fyi, temperature manipulation being so base line is why there are temperature offsets on the temperature tab, which will be applied to any temperature commands within a gcode file

ntoff, 2016-01-12:
Some slicers allow you to set an initial temperature and then another one for following layers. If you apply an offset to all of them, that could yield quite horrendous results.

Let's say you set it to 190 for PLA and turned off the ability for other layers to be a different temperature. Later you go ahead and enable it again for ABS but forget to change your start gcode so your initial temperature might still be 190. No problem you say, I'll set an offset so it's now 235 for ABS.

If there are already other temperature settings for the following layers you've now applied that offset to those which could result in a hot end temperature of 280 degrees.

Ouch.

foosel, 2016-01-12:
The feature is meant for temperature fine-tuning in existing, already sliced files (e.g. "oops, that spool of PLA needs 220 instead of 210 to print perfect, let's try +10 here then"), not for material switching ("let's print this in PEEK instead of PLA").

I understand your example scenario, but it's not the responsibility of OctoPrint to fix broken files, and such a GCODE file like you describe it I'd deem broken beyond repair. Sounds like some serious issues are bound to happen with such a range, whether an offset is active or not, either the first layer will clog up the nozzle since the temperature is too low or the following layers will clog it since the filament carbonizes. I can't hold the user's hand everywhere - operating a 3D printer is something that demands a bit of taking care of stuff and using your head to think, that also involves knowing what you are doing with your slicer and whether it would be a good idea to apply an offset here or not. It's always a very tricky walk on egg shells, trying to prevent users from shooting themselves in the foot without taking functionality away that others can't live without. Either way, someone is going to complain ;)

The OP motivated the need for a GCODE editor (which again, I agree would be a neat feature, but I'm not the only one who can write that) by stating that that would make it possible to fine tune temperatures (or at least that's how I read that). And for that the offset functionality is an already available method to achieve it.

nophead, 2016-01-12:
This is an example where RepRap went down totally the wrong route, IMHO. Slicers do too much and hosts do to little. Hydraraptor splits the concerns in a different way. I used Skeinforge only to generate the tool path. I ignore all the speed information, flow information and have no startup / end code. Since I slicer centred on 0,0 that makes my gcode machine and material independent. I only need to re-slice when I want a different layer height or infill density, etc. 

My host is responsible for controlling the machine. I give it a machine description and a material description. It controls start up / temperatures / feed speeds / flow rates / retract speeds, etc. This means I can for example print a file I sliced for hydraraptor many years ago and printed in PLA and run it my Sturdy90 and print it in ABS. Completely different machines, bed size and materials, speeds, extruder gearing, etc. The only limitation is the filament path size needs to be compatible with the nozzle aperture.

My host is just Python so it runs on any machine but it has no user interface. That should be a third component in my opinion. So slicers should do less, hosts should do more and be separated into machine control and GUI components.

foosel, 2016-01-12:
@nophead 

> This is an example where RepRap went down totally the wrong route, IMHO. Slicers do too much and hosts do to little.

Fully agree with you there. Would even go as far as to say "Slicers and firmware does too much and hosts do too little" - the firmware doesn't give me full control of the machine, the slicer doesn't give me full control of the file, and I'm stuck in the middle ;)

nophead, 2016-01-12:
Yes I should also add Hydraraptor's firmware is tiny and never changes. It only does the real time bits, which are motor stepping and heater control. My host does the acceleration.

InoSiX, 2016-01-12:
TL:DR > I agree we do not need to hold the user's hand. This function is fairly primitive and adds functionality that, while "advanced", is easy to learn (Start/End GCode uses easily understood headers prior to the values to be edited).

I have only just started with my 3D printer. While I am not attempting to stoke flames here, Foosel, you yourself said in this thread two opposite ends of the spectrum. On one hand you say a 3D printer is a big boy toy and big boys should know how to make it work without hand holding. On the other hand you say GCode editing is not for everyone because stuff can break, therefore it could be better handled in plugin. 

I think Octoprint is a incredibly neat tool. I now can leave my printer where ever, using a wireless interface to control it, connecting via WLAN. I don't need to keep it close by, with the smell and sound it creates, and even on the go while I am not home. It's also an open source project, running on open source hardware running an open source OS. I think we insult the user by suggesting they can't have access to something like Start/End GCode editing. 

I know very little about Start/End GCode, but it isn't that difficult to read about. Already, when I forget to change heated bed temps, because I don't know exactly where it is in RepG, I just edit the GCode and export again. I have failed to do this several times actually, and that is when I realized Octoprint does not have GCode editing, requiring me to either export/import again, or to delete and reupload. Does it work? Yes. Could it work better? Yes. Is it difficult to implement? I am familiar with OOP (Java specifically). Based on the functionality we have with the current GCode viewer, it seems to me like it would be incredibly easy to implement.

I can understand the position that is "the more functionality we include out of the box, the more the main team needs to maintain". But the likelihood that a plugin developer steps away from his project is a lot more likely than everyone abandoning Octoprint as a whole. Further, if something changes to the point where Octoprint can not handle GCode editing, this is likely due to a massive change in how GCode is handled across the board, including to the printer itself. Seems like a develop and forget feature, IMHO.

foosel, 2016-01-12:
@InoSiX please stop assuming anyone is saying there should not be an editor. I have no idea where you are getting this from. I thought we had it cleared up yesterday already that there is nothing that speaks against it. All I'm saying is, I won't make it a core feature, since the goal with ongoing OctoPrint development is to modularize stuff and extract things out of it instead of bunching everything together even further. You might not have noticed, but the Software Updater, the Discovery functionality, heck, even the Plugin Manager itself are plugins for this very reason.

I don't have time to work on this at the moment. Neither does Salandora. There are way more pressing issues to tackle at the moment. So if people want this feature NOW, someone else will need to develop it - but that's the beauty of the plugin system, that is something that is *possible*. And if nobody steps up and one of us devs does it, it will be a (possibly bundled) plugin in any case. And if it was now developed by a 3rd party developer, and that developer would step away - well, I guess a solution could be found if necessary. Either by someone adopting the development, or it just staying as is - as you said, it is a somewhat easy to develop and maintain feature, so a potential external plugin author also shouldn't have much overhead with it.

To be honest, I'm not entirely sure why you are getting so worked up over in what form this kind of functionality would be developed and by whom and want to keep making your point again and again (this is a feature very dear to your heart and you want to see it in OctoPrint, the earlier the better, we get it). If you are unhappy with having to wait - well, you said you even have a programming background, time to maybe take a look at the [Getting started guide](http://docs.octoprint.org/en/master/plugins/gettingstarted.html) and give it a try yourself? :) And I'm being serious here. It IS in fact a very good example for getting one's feet wet with adding functionality to OctoPrint through plugins, and it would be a nice way to give back to a project you seem to like so far.

InoSiX, 2016-01-12:
I will give it a shot. Thank you for taking the time to respond.

foosel, 2017-01-27:
This turns out to actually be a duplicate of #231

closed 2017-01-27 by foosel